PollyNET/Pollynet_Processing_Chain #29: plots

HolgerPollyNet, 2020-04-17:
For some plots the caption legend etc is not well behaving (overlaying, too big, too large). Check each output and modify.

ZPYin, 2020-05-15:
Since there are quite a lot of figures, I would recommend to collect here with the figures that need to be improved. And it's better to explain which part needs to be improved. 

Below is an example:

[image]

HolgerPollyNet, 2020-05-15:
[image]

ZPYin, 2020-05-16:
[image]

ZPYin, 2020-05-16:
[image]

ZPYin, 2020-05-16:
[image]

closed 2020-05-18 by ZPYin

HolgerPollyNet, 2020-05-18:
Ey, maybe I still need some time more ;-)

ZPYin, 2020-05-18:
The figure issues raised above have been fixed in latest commits. If you find new issues, you can re-open this topic. 😃

PR #60 (2020-05-16): Zhenping
- Improve figures (close #29)
- Add new keyword of 'tTwilight' to facilitate the water vapor calibration (#59 )
commit 7c18eb19 (2020-05-16): improvement: import figure layout

import the figure layout to fix issues raised in #29 (close #29)
commit 7066ebad (2020-05-16): improvement: improve figure layout

import the figure layout to fix issues raised in #29 (close #29)
commit 1424523c (2020-05-17): improvement: improve figure layout

import the figure layout to fix issues raised in #29 (close #29)