PollyNET/Pollynet_Processing_Chain #39: ## Interpolation of meteorological data

ZPYin, 2020-04-21:
Interpolate the meteorological data to smooth the steep change in the relative humidity between two measurement segments.

HolgerPollyNet, 2021-01-06:
I think @martin-rdz should give advice for this topic to be consistent with current cloudnet interpolation routines for meteorological data.

HolgerPollyNet, 2021-01-20:
Use of forecast data?

ZPYin, 2021-07-03:
Meteorological data interpolation has been implemented in Picasso v3.0.

See the MATLAB script for invoking meteorological interpolation:

```matlab
clc; close all;

projectDir = fileparts(fileparts(mfilename('fullpath')));

%% initialization
GDAS1Folder = fullfile(projectDir, 'data', 'GDAS1');
site = 'melpitz';
mTime = datenum(2019, 7, 10):datenum(0, 1, 0, 0, 1, 0):datenum(2019, 7, 11);
height = 0:30:10000;   % height. (m)

%% read data
tempRaw = loadMeteor(mTime, height, ...
    'meteorDataSource', 'gdas1', ...
    'gdas1Site', site, ...
    'flagReadLess', true, ...
    'gdas1_folder', GDAS1Folder);
tempInterp = loadMeteor(mTime, height, ...
    'meteorDataSource', 'gdas1', ...
    'gdas1Site', site, ...
    'flagReadLess', true, ...
    'method', 'linear', ...
    'gdas1_folder', GDAS1Folder);

%% data visualization
figure('Position', [0, 20, 500, 500], 'Units', 'Pixels', 'Color', 'w');

figPos = subfigPos([0.1, 0.1, 0.8, 0.85], 2, 1, 0, 0.04);

% raw temperature field
subplot('Position', figPos(1, :), 'Units', 'normalized');
p1 = pcolor(mTime, height/1e3, transpose(tempRaw)); hold on;
set(p1, 'EdgeColor', 'None');

xlim([mTime(1), mTime(end)]);
ylim([height(1), height(end)]/1e3);
caxis([-30, 30]);

colormap('jet');

xlabel('');
ylabel('Height (km)');

set(gca, 'XTick', mTime(1):datenum(0,1,0,3,0,0):mTime(end), 'XTickLabel', '', 'YMinorTick', 'on', 'Box', 'on', 'Layer', 'top', 'Linewidth', 2, 'TickDir', 'out');

cb = colorbar('Position', [figPos(1, 1) + figPos(1, 3) + 0.03, figPos(1, 2), 0.02, figPos(1, 4) - 0.1], 'Units', 'Normalized');
titleHandle = get(cb, 'Title');
set(titleHandle, 'string', 'T (K)');

text(0.1, 0.8, '(a) Raw', 'color', 'w', 'FontSize', 14, 'Units', 'Normalized');

% interpolated temperature field
subplot('Position', figPos(2, :), 'Units', 'normalized');
p1 = pcolor(mTime, height/1e3, transpose(tempInterp)); hold on;
set(p1, 'EdgeColor', 'None');

xlim([mTime(1), mTime(end)]);
ylim([height(1), height(end)]/1e3);
caxis([-30, 30]);

colormap('jet');

xlabel('Time');
ylabel('Height (km)');

set(gca, 'XTick', mTime(1):datenum(0,1,0,3,0,0):mTime(end), 'YMinorTick', 'on', 'Box', 'on', 'Layer', 'top', 'Linewidth', 2, 'TickDir', 'out');

datetick(gca, 'x', 'HH:MM', 'keepticks', 'keeplimits');

cb = colorbar('Position', [figPos(2, 1) + figPos(2, 3) + 0.03, figPos(2, 2), 0.02, figPos(2, 4) - 0.1], 'Units', 'Normalized');
titleHandle = get(cb, 'Title');
set(titleHandle, 'string', 'T (K)');

text(0.1, 0.8, '(b) Interpolated', 'color', 'w', 'FontSize', 14, 'Units', 'Normalized');

export_fig(gcf, fullfile(projectDir, 'test', 'comparison_meteorological_interpolation_Issue_39.png'), '-r300');
``` 

Below is the comparison before/after data interpolation:

[image]

It was disabled in Picasso v3.0 by default. But can be easily switched on by setting interpoaltion **mode** to `linear` (or `cubic`)
https://github.com/PollyNET/Pollynet_Processing_Chain/blob/4eede9655dfb04186d2d1b7ddc963ac7c20ca868/lib/io/loadMeteor.m#L66

closed 2021-07-03 by ZPYin

HolgerPollyNet, 2021-07-08:
@martin-rdz do you like it?