PollyNET/Pollynet_Processing_Chain #42: guidelines how to install sql lite

HolgerPollyNet, 2020-04-22:
just install ?

ZPYin, 2020-04-22:
You don't need to install anything for accessing SQLite, because I attached a java [SQLite driver](https://github.com/PollyNET/Pollynet_Processing_Chain/blob/master/include/sqlite-jdbc-3.30.1.jar) along with Picasso. This can help Picasso to read/write SQLite DB.

But maybe a GUI for SQLite is convenient for checking the DB file, manually. Most mainstream SQL GUI supports SQLite. 
But if you don't have such one, I would recommend a free software –– [SQLite studio](https://sqlitestudio.pl/index.rvt?act=download).

ZPYin, 2020-04-23:
Error for using the JDBC

```text
[2020-04-22 11:53:24] Start to save calibration results.
Error using database.jdbc.connection/set (line 14)
Invalid connection.

Error in save_depolconst (line 31)
set(conn, 'AutoCommit', 'off');

Error in pollynet_processing_chain_pollyxt_tropos (line 330)
    save_depolconst(dbFile, ...

Error in pollynet_processing_chain_main (line 197)
    [reportTmp] = eval(sprintf('%s(taskInfo, pollyConfig);', ...

and before warning:
Water vapor measurements were performed during daytime during 20200412 05:00 to 05:59.
Warning: No suitable driver found for
jdbc:sqlite:C:\Users\baars\Desktop\polly_try\Pollynet_Processing_Chain\results\PollyXT_TROPOS\pollyxt_tropos_calibration.db
> In load_wvconst (line 106)
  In select_wvconst (line 101)
  In pollynet_processing_chain_pollyxt_tropos (line 252)
  In pollynet_processing_chain_main (line 197) 
```

ZPYin, 2020-04-23:
Did you restart the MATLAB after the first run of the latest commit?

This problem seems to be related with the failure of importing SQLite JDBC (java database connector).

The script of `addSQLiteDriver.m` will add the  to the MATLAB JAVA search path in the first run automatically, but it requires to restart MATLAB to make MATLAB initialized with the SQLite JDBC.

There should be a prompt-up message for that, maybe you neglected it.

Restart the MATLAB and run Picasso again. 

If there is still error, please send me together with the log file. Then I can try to debug it.

closed 2020-04-23 by ZPYin