PollyNET/Pollynet_Processing_Chain #54: error analysis

ZPYin, 2020-05-05:
Implement error analysis based on error propagation, because EARLINET data format requires uncertainty for backscatter, extinction and depol profiles.

> Monte-carlo method is simpler to be implemented. But it's unrealistic for near-realtime analysis, when taking into account of the huge calculations.

HolgerPollyNet, 2020-08-06:
Hi ZPYin. Is there any advance in the error analyis? Or is there already something implemented?
If no I think I would like sto strart working on it. For the extinction we can use the error of the linear fit to estimate the uncertainty. For backscatter, I think the best is to use Monte carlo. probably not in near-real time but for profiles with GDAS.
For particle depol, we can then use the eroro propagation, for backscatter and volume depol noise? What do you think? Should we consider  the SNR as well?

ZPYin, 2020-08-06:
I have some old functions for error analysis. which are based on monte-carlo simulations. See `.\lib\polly_raman_ext_std.m` and `.\lib\polly_raman_bsc_std.m`.

These functions can be easily incorporated.

With regard to particle depol, I should also have a matlab program (based on monte-carlo) but I need to check.

All in all, I think it's not a simple task. We may need to discuss more, instead of just implementing some functions and make it work. Below are some issues that should be evaluated first:

- time requirement for error analysis
- products that need to be error analyzed
- factors that should be considered in error analysis
- intercomparisons with LabView outputs
- documentations

As this is not a public issue, I would suggest moving the discussion to email.

commit 95f61d85 (2021-07-18): Improvement: error analysis

Add error analysis for profiles.

close #54
closed 2021-07-18 by ZPYin