PollyNET/Pollynet_Processing_Chain #73: introduce old labview color scale

HolgerPollyNet, 2020-10-16:
as option

ZPYin, 2020-10-16:
Do you have the colormap in RGB?

ZPYin, 2020-10-16:
**Labview** style colormap was implemented (see the example below).

**Range-corrected signal**
[image]

**volume depolarization ratio**
[image]

The `labview` style colormap can be configured by setting `colormap_basic: "labview"` in the polly config file.

closed 2020-10-16 by ZPYin

HolgerPollyNet, 2020-10-16:
WOW, you are so cool!!!!!
    
    Thanks!
    Am 16.10.2020 um 11:04 schrieb ZPYin:
    
    
      
      Labview style colormap was implemented (see
        the example below).
      Range-corrected signal
        
      volume depolarization ratio
        
      The labview style colormap can be configured by
        setting colormap_basic: "labview" in the polly
        config file.
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/PollyNET/Pollynet_Processing_Chain/issues/73#issuecomment-709923858",
"url": "https://github.com/PollyNET/Pollynet_Processing_Chain/issues/73#issuecomment-709923858",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]
    
    
*****************************************************
 Dr. Holger Baars (Dipl.-Met.)
 Leibniz Institute for Tropospheric Research (TROPOS)
 Permoserstr. 15
 04318 Leipzig, Germany
 phone: ++49 341 2717 7314
 fax: ++49 341 2717 99 7314
 email: baars@tropos.de
 internet: http://www.tropos.de
 ****************************************************

commit 4a77fe45 (2020-10-16): Improvement: add labview colormap

Implement labview-style color map, which can be configured by setting
`colormap_basic` to `labview`.

close #73.
commit 0b6cbed4 (2020-10-29): Merge `holger` into `dev`

* Bugfixes: fixed depol-cali time (#71)

Try to merge the latest commits in `holger` branch, although it seems rather dangerous to do so... 💣 💣 💣 

But let's see what mysterious bugs will show up...

* Improvement: depol-cali res for longterm plot

Add polarization calibration results at 532 nm for pollyxt_ift long-term
calibration plot.

close #72.

* Bugfixes: incompatible dims for pcolormesh

Adjust the dims of X, Y for `pcolormesh` in housekeeping plot to be
compatible with Z.

Related issues: #70, #60
close #74.

* Improvement: add labview colormap

Implement la [... cut]