PollyNET/Pollynet_Processing_Chain #76: Change low color in labview color scale

HolgerPollyNet, 2020-10-27:
The color outside the given range seems to be white in both cases, below the color scale and above the color scale. This leads to the effect that no sinal is shown as white the same color as for clouds: 
[image]
I propose to use black as low color, i.e. in the example everything below 0 and white as high color, i.e. everything above 1.

ZPYin, 2020-10-27:
The colormap of `labview` has be adjusted. 

PS:

The figure for attenuated backscatter is generated with addtional smoothing in horizontal and vertical. And mask of low SNR was also implemented. The low-SNR bits will be colorized by the color for bad bits, which is defined as below:

https://github.com/PollyNET/Pollynet_Processing_Chain/blob/b0b749a478b1a723ffc1ec24165d216feaea9eb7/lib/python_colormap.py#L582

closed 2020-10-27 by ZPYin

ZPYin, 2020-10-27:
**Before** (white for overflow and bad bits):
[image]

**After** (white for overflow and black for bad bits):
[image]

commit b0b749a4 (2020-10-27): Improvement: adjust colormap

Set the color for bad bit in `labview` colormap to be black.

close #76
commit 0b6cbed4 (2020-10-29): Merge `holger` into `dev`

* Bugfixes: fixed depol-cali time (#71)

Try to merge the latest commits in `holger` branch, although it seems rather dangerous to do so... 💣 💣 💣 

But let's see what mysterious bugs will show up...

* Improvement: depol-cali res for longterm plot

Add polarization calibration results at 532 nm for pollyxt_ift long-term
calibration plot.

close #72.

* Bugfixes: incompatible dims for pcolormesh

Adjust the dims of X, Y for `pcolormesh` in housekeeping plot to be
compatible with Z.

Related issues: #70, #60
close #74.

* Improvement: add labview colormap

Implement la [... cut]