PollyNET/Pollynet_Processing_Chain #78: Put water marks on Pictures

HolgerPollyNet, 2020-11-03:
To avoid misuse of the images shown online and also to keep the right of the data owners, it was suggested to implement water marks in the plots generated by the processing chain.

ZPYin, 2020-11-03:
Do you have any idea in terms of the content of the water marks? Should it be identical or variant depending on the stations? Should it be text only or image?

Text water marks, see: https://matplotlib.org/3.1.0/gallery/text_labels_and_annotations/watermark_text.html
Image water marks, see: https://matplotlib.org/3.1.0/gallery/images_contours_and_fields/watermark_image.html

HolgerPollyNet, 2020-11-03:
ideal, would be per lidar. I.e. one tropos logo and one for the hosting insitition (e.g. FMI)

HolgerPollyNet, 2020-11-03:
and some text with something like: "development version of processing chain", or "beta results" or so

rico-hengst, 2020-11-03:
Hi,

we have to ask, what is our key concern?

* add logo of the data (image) provider or add logo of the observer 
institution OR both of them

* should we add also some text info?

    * provisional data

    * copyright info CC-BY-SA

ZPYin, 2020-11-04:
It would be the best to have some examples to start with. @HolgerPollyNet 

Then we can store all the available water marks, let's say, in the [`img`](https://github.com/PollyNET/Pollynet_Processing_Chain/tree/master/img) folder. I think water marks are not something confidential, correct me if it is.... 
Finally, we can implement the water marks in the `Pollynet_Processing_Chain` (might be easiler) and maybe one new `keyword` to control such behavior.

rico-hengst, 2020-11-04:
Let's discuss what kind of information we want add as watermark.
Here one suggestion, maybe to much information on it?
[image]

HolgerPollyNet, 2020-11-04:
Yes we can discuss this next week.
    
    Licence symbol, TROPOS logo, other institution logo could be merged
    in one image and added.
    
    One could also write TROPOS and the partner institution in the
    caption simply, and just add the licence logo and a watermark for
    polly.tropos.de covering the entire image ;-)
    
    We should simply start and try, maybe with Rico's proposal. And then
    ask for feedback ;-)
    
    Best
    Holger
    
    
    
    Am 04.11.2020 um 11:13 schrieb
      rico-hengst:
    
    
      
      Let's discuss what kind of information we want add as
        watermark.
        Here one suggestion, maybe to much information on it?
        
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/PollyNET/Pollynet_Processing_Chain/issues/78#issuecomment-721641090",
"url": "https://github.com/PollyNET/Pollynet_Processing_Chain/issues/78#issuecomment-721641090",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]
    
    
*****************************************************
 Dr. Holger Baars (Dipl.-Met.)
 Leibniz Institute for Tropospheric Research (TROPOS)
 Permoserstr. 15
 04318 Leipzig, Germany
 phone: ++49 341 2717 7314
 fax: ++49 341 2717 99 7314
 email: baars@tropos.de
 internet: http://www.tropos.de
 ****************************************************

martin-rdz, 2020-11-04:
> 
> 
> Let's discuss what kind of information we want add as watermark.
> Here one suggestion, maybe to much information on it?
> [image]

That modification is not covered by the Tropos style guide [I guess we should obey it].

An example, with cc-by and a text can be found here: https://github.com/martin-rdz/global_gb_cloud_aerosol_profiling

Python code for including figures is pretty simple (example from the [jupyter notebook](https://github.com/martin-rdz/global_gb_cloud_aerosol_profiling/blob/master/overview_g-b_profiling.ipynb), cannot tag the lines there)
```python
im = matplotlib.image.imread('cc-botton-by-sa.png')
newax = fig.add_axes([0.812, 0.16, 0.2, 0.18], zorder=3)
newax.imshow(np.rot90(im))
newax.axis('off')

ax.text(1.0175, 0.515, 'Data: Jan 2019', fontsize=10, 
        bbox=dict(facecolor='none', alpha=0.85, edgecolor='none'), transform=ax.transAxes, 
        rotation=90, verticalalignment='baseline', horizontalalignment='center')
# or
# ax.text(0.6, 0.02, 'Data: Jan 2018; Visualisation: @martin_ra_', fontsize=10, 
#         bbox=dict(facecolor='white', alpha=0.7), transform=ax.transAxes)
```
Rotation is optional. I would just add the cc-by logo, then a Text string with Tropos + other institutions and something like 'preliminary results', 'experimental retrieval' or 'retrieval under development' below the plot.

I would advise again a watermark overlying the data (as in the matplotib example), because 

1. we frequently use the plots on our own. dealing with a data covering watermark would be annoying.
2. if someone removes this statement, that person is willingly disobeying the data policy (if not acknowledging us otherwise).

HolgerPollyNet, 2020-11-10:
We agree with Martins suggestions to:
1. we frequently use the plots on our own. dealing with a data covering watermark would be annoying.
2 .  if someone removes this statement, that person is willingly disobeying the data policy (if not acknowledging us otherwise).

So lets add somethng like this:
https://github.com/martin-rdz/global_gb_cloud_aerosol_profiling

to the plots, maybe at the bottom or at the side.

I propsoe, depending on the lidar, its the used lcinece logo, the tropos logo (ot text), the partner logo (or text), and the hint for preliminary data.

ZPYin, 2020-11-12:
See below an example after adding watermark. All watermarks associated with copyright are located on the edge of the plot (but may be different due to different layouts). Institutional marks are highlighted with text, with a leading copyright symbol. (I tried logo, but seems a little bit distracting...). 

Any comments?

**Example for watermark**
[image]

PS,

If you have more `wonderful` ideas in terms of watermark, I would suggest to wait for Picasso v3.0. The framework of Picasso v2.0 is not friendly for adding new data visualization features. 🤣

HolgerPollyNet, 2020-11-12:
Hey, that looks in principle great and sufficient, except that a
    copyright statement is not needed when we have the CC licence (this
    is a contradiction).
    
    Rico, would "licensed by TROPOS and (UMA/FMI/DWD)" be correct?
    
    Best
    Holger
    
    Am 12.11.2020 um 03:48 schrieb Zhenping
      Yin:
    
    
      
      See below an example after adding watermark. All watermarks
        associated with copyright are located on the edge of the plot
        (but may be different due to different layouts). Institutional
        marks are highlighted with text, with a leading copyright
        symbol. (I tried logo, but seems a little bit distracting...).
      Any comments?
      Example for watermark
        
      PS,
      If you have more wonderful ideas in terms of
        watermark, I would suggest to wait for Picasso v3.0. The
        framework of Picasso v2.0 is not friendly for adding new data
        visualization features. 🤣
      —
      [

{

"@type": "EmailMessage",

"@type": "ViewAction",

"target": "https://github.com/PollyNET/Pollynet_Processing_Chain/issues/78#issuecomment-725794026",

"url": "https://github.com/PollyNET/Pollynet_Processing_Chain/issues/78#issuecomment-725794026",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"@type": "Organization",

"name": "GitHub",

"url": "https://github.com"

}

}

]
    
    
*****************************************************
 Dr. Holger Baars (Dipl.-Met.)
 Leibniz Institute for Tropospheric Research (TROPOS)
 Permoserstr. 15
 04318 Leipzig, Germany
 phone: ++49 341 2717 7314
 fax: ++49 341 2717 99 7314
 email: baars@tropos.de
 internet: http://www.tropos.de
 ****************************************************

HolgerPollyNet, 2020-11-20:
Thanks  a lot! It works well except for the partner label: 
http://picasso.tropos.de/archive_link/recent_plots_new/PollyXT_DWD/2020/11/20/2020_11_20_Fri_DWD_06_00_01_TC.png

ZPYin, 2020-11-21:
> Thanks a lot! It works well except for the partner label:
> http://picasso.tropos.de/archive_link/recent_plots_new/PollyXT_DWD/2020/11/20/2020_11_20_Fri_DWD_06_00_01_TC.png
> 
> In the respective config it is written: "partnerLabel": "DWD",

I developed this feature with using partnerlabel array (eg, {"UMA", "DWD"}), but the final commit is based on parnerlabel string (eg, "UMA"). Therefore, one additional indexing is used, which create the output of the first character of parnerlabel.

It has been fixed in the latest commit. (Already pulled to rsd2.)

ZPYin, 2020-11-23:
Second drafted version: 
[image]

The layout of the text may not look good, but better than nothing. I will re-organise it when I start Picasso v3.0.

HolgerPollyNet, 2020-11-23:
Perfect. We just have to find out if we use 3.0 or 4.0.

@rico-hengst can you give advise?

HolgerPollyNet, 2020-11-23:
ok 4.0. everything fine! The (c) can stay according to Copenricus.

HolgerPollyNet, 2021-02-12:
Little bug, if no Partner is existent, the "and" symbol should not appear:
[image]

Furthermore, the Numbers on the right axis seem to be cut off  a little bit. Thus, label size should be decreased or plot size increased.

ZPYin, 2021-02-14:
It has been fixed. see the image below:

[image].

closed 2021-02-14 by ZPYin