PollyNET/Pollynet_Processing_Chain #79: Implement storing in earlinet netcdf format

HolgerPollyNet, 2020-11-03:
A script, converting the Picasso output to Earlinet DB format was already created. Now, this script should be implemented in the processing chain, to automatically create this files using the detected lowest cloud base as top height and the height of full overlap (or something else) as lower limit.

HolgerPollyNet, 2021-08-31:
New EARLINET under development - maybe wait until that. Or leave conversion script as stand alone --> to be dicussed.

HolgerPollyNet, 2023-02-07:
I hope we can reactivate this issue in near future. Maybe together with or after issue  #180

ZPYin, 2023-02-07:
I don't suggest to incorporate [polly2earlinetDB](https://github.com/PollyNET/polly2earlinetDB) into Pollynet Processing Chain at the moment. Two reasons:
- [polly2earlinetDB](https://github.com/PollyNET/polly2earlinetDB) is self-complete and easy to be managed and mantained as a little small project
- To incorporate [polly2earlinetDB](https://github.com/PollyNET/polly2earlinetDB) to Pollynet Processing Chain, we will need to be adapt to EARLINET data format changes. Each time that EARLINET data format changes, we will have to update Pollynet Processing Chain. This will bring additional version changes for Pollynet Processing Chain.

As far as I can see, there is no barrier to run Pollynet Processing Chain and polly2earlinetDB in parallel or in different cronjobs at the moment. (Correct me if not...)

Anyway, I'm open to hear more about your ideas. If we find it's really necessary to incorporate polly2earlinetDB into Pollynet Processing Chain for some reason, we can also discuss how to achieve that.