PollyNET/Pollynet_Processing_Chain #91: raw temporal resolution

HolgerPollyNet, 2021-01-12:
I still have the feeling, that a fixed temporal resolution of 30 seconds is assumed. However, in practise it is between 30 and 31 seconds. This leads to wrong x-axis in the color plots ending too early:
[image]

in contrast the old verlauf plot:
[image]

Suggestion: Use start and end time and the number of profiles to calculate the mean delta_t, if a fixed one is needed:delta_t_mean= (end-start)/N_profiles

ZPYin, 2021-02-14:
It has been fixed in Picasso v2.0. But I will keep it open to remind myself of this issue in the designing phase of Picasso v3.0.

ZPYin, 2021-07-03:
The temporal resolution now can be set to be 30 s by setting keyword of `flagCorrectFalseMShots` to `true`.
https://github.com/PollyNET/Pollynet_Processing_Chain/blob/59b2f52611a69657be1ab380bb21d53dd86c2729/lib/config/polly_global_config.json#L2

By default, Picasso read measurement time from netCDF file, which could be not in perfect 30 s interval due to issues in data recording system. This would lead to problematic timestamps as listed above.

But sometimes, maybe the problematic timestamp can also be helpful to monitor the data recording system (who knows...). I leave the option of correcting the timestamp to the users.

closed 2021-07-03 by ZPYin