PollyNET/Pollynet_Processing_Chain #92: temperature correction for Raman channles

HolgerPollyNet, 2021-01-15:
We realized, that in some Polly systems there is a temperature effect in the Raman channels due to the very narrow bandwidth of the interference filters.

For that systems we need a correction based on atmospheric temperature. We do not have this correction function yet, but the framework for a correction as a function of T should already be implemented. The function would be applied on the raw signal:

This is the current experimental function:

C=(exp(-0.001*T)/exp(-0.001*210));
Praw_corr=Praw/C;

HolgerPollyNet, 2021-01-20:
need to clarify, if raw signal need to be background (dead-time) corrected first

ZPYin, 2021-07-06:
This functionality has been implemented in Picasso v3.0. Please let me know which polly data can be used for testing.

⚠️ ⚠️ ⚠️ 
> It's assumed only one type of temperature correction formula is required (, although I guess it's theoritically bandwidth dependent).

HolgerPollyNet, 2021-07-06:
We will provide an example asap! Stay tuned!

closed 2021-07-13 by ZPYin

ZPYin, 2021-08-03:
Matlab symbolic functions (string formulas) have been implemented for signal temperature correction. The correction functions can be specified in polly configuration file as strings, see the example below.

```json
"tempCorFunc": ["1", "1", "exp(-0.005 * T)", "1", "1", "1", "exp(-0.001*T - 0.617)", "1", "1", "1", "1", "1", "1"],
```

This will correct Raman signal at 387 and 607 nm with exponential functions, in which the temperarure (in Kelvin) can be loaded from GDAS1 or other meteorlogical database.

HolgerPollyNet, 2022-05-31:
@gast-ben: Here you see the implementation for correction function.

commit 642a0585 (2021-07-13): Enhancement: temperature correction

Implement temperature correction for Raman channels.

close #92.