PollyNET/Pollynet_Processing_Chain #94: effective resolution for lidar ratio

HolgerPollyNet, 2021-02-12:
Should we provide the lidar ratio (Raman method) with the effective resolution? That would mean to take different smoothing lengths for extinction and backscatter coefficient.

I think this could be implemented as an option in the future.

ZPYin, 2021-02-14:
I will take this into account in designing Picasso v3.0. In principle, it should be easy to be implemented. 
The headache thing is to homonize the effective resolution of backscatter and extinction profiles. But of course, this needs to be addressed in Picasso v3.0.

ZPYin, 2021-07-20:
Smoothing window of extinction and backscatter can be harmonized by using additional 2-order Savitzky-Golay filter for backscatter, when effective resolution of backscatter is less than that of extinction. The smooth window length of the filter can be determined by Eq.6 in Ref.1.

This feature has been implemented in Picasso v3.0.

**References**
1. Mattis, I., D'Amico, G., Baars, H., Amodeo, A., Madonna, F., and Iarlori, M.: EARLINET Single Calculus Chain–technical–Part 2: Calculation of optical products, Atmospheric Measurement Techniques, 9, 3009-3029, 2016.

commit 249180b1 (2021-07-20): Enhancement: eff res lidar ratio

Harmonize effective resolution of particle backscatter and extinctio for
lidar ratio calculations.

Close #94
closed 2021-07-20 by ZPYin