PollyNET/Pollynet_Processing_Chain #95: How to Licence the PollyNET output

HolgerPollyNet, 2021-02-12:
**Moritz:**
I've just checked global attributes of the netcdf files from the lidar products of the Max Planck Institute. There, a copyright statement is included "The copyright for these data is with the Max-Planck-Institut fuer Meteorologie, Hamburg. Any use is subject to written consent of the copyright owner." Do we need anything similar? Furthermore, I found "DetectionMode = "Photoncounting". All Pollies work in photocounting mode, but maybe it could be mentioned.

**TROPOS:**
_Authorship and rights of use_ 
The author of the research data can restrict its use and exploitation. The research data of TROPOS should be made available to third party users, whereby in this context a contribution to the scientific reputation of the author at TROPOS can also
be made. In order to ensure a high usability of the research data and to promote the reputation of the author, the research data of TROPOS should be labelled with the Creative Commons licence CC-BY SA 3.0, which 
• allows the free use of the research data also for commercial purposes, 
• users agree to cite the authors of the research data and
• allows the publication of user-generated derivatives of this research data only
under the same licence conditions.
Research data may be labelled with a different licence if the research data was collected in the context of a scientific project and this licence has been agreed.

**Holger:** 
How do the other (institutes) think?
Any problem using CC-BY SA licence?

HolgerPollyNet, 2021-05-04:
I added now Licence CC-BY-SA 4.0 in the netcdf and the PollyNET data policy as on the website. Nevertheless, I realized, it would be good to have the PI Name in the locals configs, so that this can be stored in the netcdf as well.

ZPYin, 2021-07-04:
It has been implemented in Picasso v3.0.

The information of PI and data originator can be configured in polly configuration file (see example below):

```
    "PI": "Holger Baars",
    "PI_affiliation": "",
    "PI_affiliation_acronym": "TROPOS",
    "PI_address": "",
    "PI_phone": "",
    "PI_email": "",
    "Data_Originator": "",
    "Data_Originator_affiliation": "",
    "Data_Originator_affiliation_acronym": "",
    "Data_Originator_address": "",
    "Data_Originator_phone": "",
    "Data_Originator_email": "",
    "comment": "",
```

commit c8f8f37d (2021-07-04): Improvement: add PI info to nc

Add PI information to netCDF files. The PI information can be configured
in polly configuration file.

Available attributes include:
- PI
- PI_affiliation
- PI_affiliation_acronym
- PI_address
- PI_phone
- PI_email
- Data_Originator
- Data_Originator_affiliation
- Data_Originator_affiliation_acronym
- Data_Originator_address
- Data_Originator_phone
- Data_Originator_email
- comment

close #95
closed 2021-07-04 by ZPYin