PollyNET/Pollynet_Processing_Chain #97: Proposal for depol calibration: use and store eta instead of V*

HolgerPollyNet, 2021-02-19:
In the frame of Picasso 3.0 I would propose to not calculate and store any longer the V* but simply the eta.

The one need to calculate V* during the depol calculation, but with this approach the calibration is independent of the used transmission ratios. Furthermore, the implementation of the GHK procedure would be straight forward without the need for a reprocessing of calibration measurements. 

eta=SQRT([+45_s/+45_g]*[-45_s/-45_g])
while
v*=((1+Dg)/(1+Ds))*eta.

In this way, the long term stability of the calibration could be also easier monitored,

@Moritz-TROPOS formulas correct?

Moritz-TROPOS, 2021-02-19:
Yes, the formulas are correct.
The GHK formalism needs just the calibration constant eta and no longer V*.

ZPYin, 2021-07-06:
Eta from polarization calibration has been implemented. Polarization calibration constant (V*) was still kept in Picasso v3.0. But it has been replaced with eta in all respective plots and netCDF files, i.e., long-term calibration plot.

Due to the discard of V*, `polly_default*.json` needs to be changed as well. New keywords of `polCaliEta{wavelength}` and `polCaliEtaStd{wavelength}` were added to configure default eta-s. See below: 

https://github.com/PollyNET/Pollynet_Processing_Chain/blob/e44ee7177ae42cc04c0e17d7329d8b3a5bd8f6d7/lib/config/polly_global_defaults.json#L2-L9

Old keywords of `depolCaliConst{wavelength}` and `depolCaliConstStd{wavelength}` was still kept, but with no actual effects on Picasso.

After pull this commit to your local machine or to the server, all polly default files should be modified to add the new keywords. But it will still keep running without any modification, as the default values of `polCaliEta` and `polCaliEtaStd` can be inherited from [`polly_global_default.json`](https://github.com/PollyNET/Pollynet_Processing_Chain/blob/v3.0/lib/config/polly_global_defaults.json).  [`polly_global_default.json`](https://github.com/PollyNET/Pollynet_Processing_Chain/blob/v3.0/lib/config/polly_global_defaults.json) is a new feature for Picasso v3.0, which is silimar to `polly_global_config.json` file. It serves as a template for all polly defaults.

closed 2021-07-06 by ZPYin

HolgerPollyNet, 2021-07-07:
@Moritz-TROPOS you will like it.

commit e44ee717 (2021-07-06): Enhancement: implemente eta

Replace polarization calibration constant V* with eta. Save eta in
netCDF file instead of V* and diplay eta in long-term calibration plot.

This commit is based on issue #97.

Close #97