PollyNET/Pollynet_Processing_Chain #106: Varying lasershot repetition rates

Ropzn, 2021-07-02:
Concerning
 Pollynet_Processing_Chain/lib/pre-process/pollyPreprocess.m   in v3.0

Maybe I don't understand it right. 
Is the strategy right now, that we a assume the laser repetition rate to be constant (beeing our reference clock) and accordingly brute force reset the timestamps, as soon as it's not 600 shots/30sec ?

If so, than that should be the other way around, or ?

I would guess the time stamp is the reference clock and sometimes some more or fewer shots occur.
Maybe ask Ronny for clarification.

Moreover the function uses a fixed number of 600, which probably stands for 20shots/sec in 30 sec.
However the new diode-pumped systems work on 100shots/sec.
So this number somehow should be replaced by a variable.

`%% Re-sample the temporal grid to 30 s, if it's not in 30s.
nInt = round(600 / nanmean(data.mShots(1, :), 2));   % number of profiles to be
                                                     % integrated. Usually, 600
                                                     % shots per 30 s`

ZPYin, 2021-07-03:
Thanks for reporting this issue. I will handle it in the next few days.

closed 2021-07-04 by ZPYin

HolgerPollyNet, 2021-07-07:
@Ropzn is this solution ok for you?

commit 092847ce (2021-07-04): Improvement: dynamic 30s mshots

Setting measurement shots for every 30 s interval according to laser
repetition rate.

close #106