PollyNET/Pollynet_Processing_Chain #116: Klett_BSC_NR not stored in done file list

HolgerPollyNet, 2021-11-05:
I realized that the information for the Klett bsc near range is not reported in the done filelist and thus not stored to the data base.

ZPYin, 2021-11-06:
This has been fixed.

## Bug Analysis

The figure name of Klett bsc near range plot was wrong when making done_filelist, see the code below:
https://github.com/PollyNET/Pollynet_Processing_Chain/blob/82bc73fea5b2eb5ffad9145b48ef574cfa25b890/lib/io/pollyWriteDonelist.m#L228-L229

thus, the figure was filtered out in writing the done_filelist.

HolgerPollyNet, 2021-11-09:
Hi @ZPYin, I and @martin-rdz  realized, that no near-range Klett backscatter is calculated at all. See for example all the recent plots at the website. Only far-range bsc is plotted. Do you have any explanation why?

ZPYin, 2021-11-09:
Regarding the near-range products, I do see both the netCDF4 files and plots in the server. Below is a sample: 

**images**
[image]

**netCDF4 files**
[image]

In addition, the plots information has been saved in the polly DB successfully as well.
[image]

So, the only explanation why they are missing at the website is that they are just not showing!!! 
To my perspective, this is something out of Pollynet Processing Chain. Please transfer this issue to Rico (@rico-hengst ).

HolgerPollyNet, 2021-11-09:
The problem is not, that it is not shown on the website, the problem is that the products are not calculated. I.e., the netcdf and images are there, bot no NR Klett product inside. So it is for sure a problem of the Processing Chain as it holds for all Pollys.

Arbitrary example of today:
[image]

While for raman there are NR products: 
[image]

Any idea where to look at?

martin-rdz, 2021-11-09:
Somehow `aerBsc_klett_NR_532` only contains masked values.

Example ``
```
:version = "3.1" ;
:history = "Last processing time at 2021-11-02 14:21:41 by pollySaveNRProfiles, 
git branch: master, git commit: 0b3e3a467de0f1f72fd979f89f7ab1a6cbe898c0" ;
data:

 aerBsc_klett_NR_532 = _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _,
    _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _,
    _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _,
    _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _,
    _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _,
    _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _,
    _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _,
    _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _,
```

Similarly for data in August 2020 (`2020_08_31_Mon_LACROS_12_00_01_1641_1759_NR_profil
es.nc`), where the overlap should have been even better.

```
:version = "3.1" ;
:history = "Last processing time at 2021-11-06 06:45:01 by pollySaveNRProfiles, 
git branch: master, git commit: 8b4e8286d58208a0b1c820b33c234a448792ca08" ;
data:

 aerBsc_klett_NR_532 = _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _,
    _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _,
    _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _,
    _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _,
    _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _, _,
```

For both instances FR and OC contains reasonable data.

ZPYin, 2021-11-10:
Thanks for the detailed information. 

This was caused by some (stupid) coding bugs. (see below)
https://github.com/PollyNET/Pollynet_Processing_Chain/blob/8b4e8286d58208a0b1c820b33c234a448792ca08/lib/interface/picassoProcV3.m#L1232-L1235
https://github.com/PollyNET/Pollynet_Processing_Chain/blob/8b4e8286d58208a0b1c820b33c234a448792ca08/lib/interface/picassoProcV3.m#L1284-L1287

The near-field results were not transferred to the right variables...

I have fixed it in the latest fixes and updated the version to `3.2` to identify the changes. Below is the new near-field plots:

[image]

Sorry for the inconvenience.

closed 2021-11-10 by ZPYin

HolgerPollyNet, 2021-11-10:
@ZPYin many thanks for the fast fixing and no worries about the bug - bug fixing is one of the major task while coding ;-)

commit 8b4e8286 (2021-11-06): Bugfixes: add Klett Bsc NR

Insert the metadata of Klett Bsc NR figure to the done_filelist.

close #116
commit 6b95383f (2021-11-10): Bugfixes: show nr Klett profiles

Fix bugs associated with missing near-field profiles with Klett method. Update the version `3.2` to identify this change.

Close #116
commit babdf247 (2021-11-25): Added near-range SNR to pollySaveNRAttnBeta plus older commits (#125)

* Bugfixes: add Klett Bsc NR

Insert the metadata of Klett Bsc NR figure to the done_filelist.

close #116

* Bugfixes: show nr Klett profiles

Fix bugs associated with missing near-field profiles with Klett method. Update the version `3.2` to identify this change.

Close #116

* Andi branch 01 (#121)

* added database param -AutoCommit- and value -off- in var. conn and commited out set(conn,...)-function

* added database param -AutoCommit- and value -off- in var. conn and commited out set(conn,...)-function continued

* a [... cut]
commit e94b7b7d (2021-12-07): Homogenization of variable names for OC (overlap corected) and NR (near-range) files (#133)

* Added near-range SNR to pollySaveNRAttnBeta plus older commits (#125)

* Bugfixes: add Klett Bsc NR

Insert the metadata of Klett Bsc NR figure to the done_filelist.

close #116

* Bugfixes: show nr Klett profiles

Fix bugs associated with missing near-field profiles with Klett method. Update the version `3.2` to identify this change.

Close #116

* Andi branch 01 (#121)

* added database param -AutoCommit- and value -off- in var. conn and commited out set(conn,...)-function

* added database param - [... cut]