PollyNET/Pollynet_Processing_Chain #117: homogenize profile output

HolgerPollyNet, 2021-11-05:
after some discussion it was decided, to make the content of the profiles, NR_profiles_ and OC_profiles files consistent. meaning same variable having same names. This means the near-range backscatter and far-range backscatter have the same variable name, but the filename were it is stored should be different as well as the meta data.

ulysses78, 2021-11-09:
That is the way I also would do it. Great!

ulysses78, 2021-12-06:
That means, that e.g.:
'NR_attenuated_backscatter_355nm' ---> 'attenuated_backscatter_355nm'

One question arrises at this point:
how to name the long_names of the variables?
Can we leave it as it is: 'near-field attenuated backscatter at 355 nm'?
Or do we have to change it to 'attenuated backscatter at 355 nm'?
In my opinion we can leave it as it is, because this is just an attribute, not a whole variable.
@HolgerPollyNet ... what do you think?

ulysses78, 2021-12-06:
This affects the following files:
- pollySaveNRAttnBeta.m
- pollySaveNRProfiles.m
- pollySaveOCAttnBeta.m
- pollySaveOCProfiles.m

HolgerPollyNet, 2021-12-06:
I think, we can leave it as the long-name is a more proper description and it would be give the chance for distinguishing.....

closed 2021-12-06 by ulysses78