PollyNET/Pollynet_Processing_Chain #119: Lower-case characters of polly-folders in level1a datastructure on rsd2

ulysses78, 2021-11-09:
### Status quo:

in level1a of polly measurements every folder name has upper- and lower-case characters 

### Goal:

in level1a every polly folder must have every folder-name with lower-case characters, as it is at the moment in level0 folder

ulysses78, 2022-01-31:
I started a trial by renaming the folder:
```
/data/level1/polly/PollyXT_TROPOS
```
to
```
/data/level1/polly/pollyxt_tropos
```

@HolgerPollyNet  ... could you please check if everything works fine in that specific folder (+subfolders) after renaming (e.g. some cron-jobs, processing-chains, ... )? Please inform me if some problems occur.

ulysses78, 2022-02-01:
I just "discovered", that Bildermacher2 creates a new folder
```
/data/level1/polly/PollyXT_TROPOS/2022
```
and puts the picasso output there :-(

HolgerPollyNet, 2022-02-01:
@ulysses78. I guess not, but do we really need the lower case? If yes, we have to change the code. But please wait because I have still a pull request pending. But today is Chinese new year! So @ZPYin will not be available!

PS: Happy new year!

ulysses78, 2022-02-03:
I have renamed the polly_tropos folder mentioned above to its original upper case format!

ZPYin, 2022-02-06:
> I just "discovered", that Bildermacher2 creates a new folder
> 
> ```
> /data/level1/polly/PollyXT_TROPOS/2022
> ```
> 
> and puts the picasso output there :-(
> 
> That finally means, if we want to change those foldernames to lower case, we have to touch the picasso matlab code! That is somehow not nice! Any other suggestions? FYI: @ZPYin

The folder name was somehow inherited from the [`pollynet_processing_chain_config_links.xlsx`](https://github.com/PollyNET/Pollynet_Processing_Chain/blob/master/config/template_pollynet_processing_chain_config_links.xlsx) file (first column `instrument`). There was no hardcoded polly names inside the matlab code.
If you want to standardize the folder name into lower-case or other formats, you can just edit the excel file. 

Let me know if this can do the trick.

ulysses78, 2022-02-16:
Thanks for this information! Depending on the campaign/location, there are multiple entries of e.g. "PollyXT_TROPOS" in the  column "instrument". Do one have to change all those entries to lower-case or only the last one or the first one or...?

HolgerPollyNet, 2022-02-16:
all

ulysses78, 2022-09-23:
No problems until now. The renaming to lower-case is working! Issue will be closed.

closed 2022-09-23 by ulysses78