PollyNET/Pollynet_Processing_Chain #120: Function "SET" has been removed in MatLab-version R2018a compared to MatLab-version 2017

ulysses78, 2021-11-12:
Since the function "SET" (set.m) has been removed in MatLab-version R2018a compared to MatLab-version 2017, the parameter 'AutoCommit' and its value 'off' has to be set in the function database.m. As far as I know, this affects the following files:
```
lib/io/extract_cali_results.m
lib/io/saveDepolConst.m
lib/io/saveLiConst.m
lib/io/saveWVConst.m
```
This was taken into account in the new branch "andi_branch_01". 
So this issue is more or less a hint, not a todo.

ulysses78, 2021-11-12:
I forgot to mention, that this new settings in "database.m" should work for MatLab-version 2017 as well!

ZPYin, 2021-11-12:
Hi, @ulysses78 

Thanks for contributing to Pollynet Processing Chain.

Please see my comment to your PR.

ulysses78, 2021-11-15:
I have now added the matlab-version check to correctly set database connection parameters in my branch and testet it with the 2018matlab-version.
@HolgerPollyNet : Please check with your 2017-version.
@ZPYin : Please check with your 2014-version

ZPYin, 2021-11-15:
> I have now added the matlab-version check to correctly set database connection parameters in my branch and testet it with the 2018matlab-version. @HolgerPollyNet : Please check with your 2017-version. @ZPYin : Please check with your 2014-version

Please see my comment to your PR (#121 ) aiming at this issue.

closed 2021-11-23 by ulysses78