PollyNET/Pollynet_Processing_Chain #123: Times of calculated overlap functions

martin-rdz, 2021-11-19:
Is the overlap function calculated at each profile?

When looking at the output, I can only find one overlap file for each 6-hour period, e.g.,
```
/data/level1a/polly/PollyXT_LACROS/2020/08/20/2020_08_20_Thu_LACROS_00_00_01_overlap.nc
/data/level1a/polly/PollyXT_LACROS/2020/08/20/2020_08_20_Thu_LACROS_06_00_01_overlap.nc
/data/level1a/polly/PollyXT_LACROS/2020/08/20/2020_08_20_Thu_LACROS_12_00_01_overlap.nc
/data/level1a/polly/PollyXT_LACROS/2020/08/20/2020_08_20_Thu_LACROS_18_00_01_overlap.nc
```
In case the overlap is calculated per profile, would it make more sense to also store it in the profile netcdfs?
Or at least in a extra netcdf similar to the profiles?

ZPYin, 2022-12-24:
Timestamps has been inserted to the overlap files. In addition, the overlap function has been converted to a 2-dimension array (time x height) with the length of the dimension of time to be 1. I've enclosed a file as an example of the new overlap file (see the link below). 

[2022_11_03_Thu_UWA_12_00_08_new_overlap.zip](https://github.com/PollyNET/Pollynet_Processing_Chain/files/10298149/2022_11_03_Thu_UWA_12_00_08_new_overlap.zip)

Hope this issue can be closed.

HolgerPollyNet, 2023-01-04:
okay, I will pull the latest version to the server, and if no error occurs we can close this issue! Many thanks!

closed 2023-01-05 by HolgerPollyNet