PollyNET/Pollynet_Processing_Chain #124: near range SNR in att bsc files

griesche, 2021-11-19:
It would be great to save also the near-range SNR in the netCDF att bsc files. So far, only the far-range SNR is stored in these files.

HolgerPollyNet, 2021-11-23:
Could this be a job for @ulysses78  ?

ulysses78, 2021-11-23:
I can do that. Should be no problem (I guess). Is this the same SNR-variable which is altready stored  in the "normal" att_bsc-files? Only for 355 and 532 nm?
Do you also want to store the quality-mask-variable (355+532nm)?

griesche, 2021-11-23:
What do you mean by 'normal att_bsc_files'? I also already started on this, but if you're faster you can also do the job :)

Yes, only 355 and 532 nm. I think the quality mask would be good as well.

ulysses78, 2021-11-23:
By 'normal att_bsc_files' I mean the att_bsc files of the far range, where those variables are already implemented.

ZPYin, 2021-11-24:
So far, the near-range `SNR` is calculated inside `Picasso`, but was not saved in the netCDF file. 
You can extract the near-range `SNR` from `data.SNR` (3D dataset: channel x height x time), and implement the **saving** within `pollySaveNRAttnBeta.m`, similar to what was done in `pollySaveAttnBeta.m` for far-range variables.

griesche, 2021-11-24:
I have updated the pollySaveNRAttnBeta.m so that the near-range SNR is also stored. I also started with the quality mask, but this is not yet in the data input to pollySaveNRAttnBeta.m. 

I have opened a pull request to the dev branch,

ulysses78, 2021-12-06:
Can this issue be closed?

griesche, 2021-12-06:
Yes I think so.

closed 2021-12-06 by ulysses78