PollyNET/Pollynet_Processing_Chain #143: Bug in Raman retrievals?

HolgerPollyNet, 2022-03-09:
Hi all,

since some time, I am wondering about some "strange" Raman backscatter retrievals at day time. At the beginning I thought this is due to background noise, but while trying to "optimize" the setting for the reference height search, I realized that I cannot reproduce the profiles.

Here is what Picasso gives for one nice example from  Mindelo on 23 July at around 9:21 to 09:46 UTC:
As you see the profiles are way to high. The reference height is 6.9 to 8.9 km and the reference value is 5e-8. **But as you see, the whole profile does not reach this value at all in this height range.** 

[image]

I calculated the same profile with my own, old program and here is how it looks like there:
[image]
I know the profiles are far from perfect, but it is the same settings as used in Picasso, and the results are quite different.

Do you have any idea where to look at? My personal feeling is, that it might be related to the transmission correction...

ZPYin, 2022-03-12:
Yes, it's a bug associated with negative signal filtering. I've fixed it in a new commit (to `dev` branch). Below are the details regarding this bug.

## Bug Description

Wrong (Raman) backscatter retrieval, where clear positive bias can be seen for 355 and 532 nm. (see the comment above)

## Bug Analysis

Negative signals were filtered when calculating the summed signal at the reference height range, both for Raman and Elastic signal. 
https://github.com/PollyNET/Pollynet_Processing_Chain/blob/be0e33589fc9aa9f32d71efd3bf506c49899d4e2/lib/retrievals/pollyRamanBsc.m#L170-L177

While Raman signal is usually weaker than Elastic signal; thus this signal filtering will make Raman go towards more positive bias. This will finally contribute to positive bias for the backscatter. In addition, when the signal gets noiser (**at daytime**), more negative values will be wiped out, Then you see much larger positive bias... 

## Bug Fixes

Remove the negative signal filter... (Actually I have done that for 1064, but forget to repeat it for 355 and 532... OK! my bad...)

Now the profile looks like more consistent with the Labview results (see below).

[image]

HolgerPollyNet, 2022-03-14:
Great! Thanks for your work! Your expert knowledge is indispensable :-). Let us check the operational data for 2-3 days and then we can close the issue!

Best
Holger

commit 36ee8e0c (2022-03-12): Bugfixes: rm negative sig filter

Negative signal filter for calculating summed signal at reference height, leads to positive bias of (Raman) backscatter. Therefore, it was removed in this commit.

close #143
PR #144 (2022-04-05): changed pollyxt_names from uppercase to lowercase
commit 1d3966521ec4f25f6224149295205b445b6f23d6 (HEAD -> dev, origin/dev)
Author: Baymax <test@email.com>
Date:   Wed Mar 16 14:13:48 2022 +0000

    changed pollyxt_names from uppercase to lowercase in config/pollynet_processing_chain_config_links.xlsx

commit 527bb5607b833d72750689270c2254d92dc04fc7
Author: Baymax <test@email.com>
Date:   Mon Mar 14 14:35:55 2022 +0000

    tracked folder \config

commit 39d1d786f2d525206ed2fcc5b2fc9e5d2f2db4d6
Author: Baymax <test@email.com>
Date:   Mon Mar 14 12:02:47 2022 +0000

    changed format from dos [... cut]
closed 2022-04-05 by ZPYin