PollyNET/Pollynet_Processing_Chain #152: hardcoded iwv format files?

HolgerPollyNet, 2022-04-12:
https://github.com/PollyNET/Pollynet_Processing_Chain/blob/0bd0d80450af4cdb781c29c7125c94e304ed5db6/lib/io/readIWV.m#L64

or https://github.com/PollyNET/Pollynet_Processing_Chain/blob/0bd0d80450af4cdb781c29c7125c94e304ed5db6/lib/io/readIWV.m#L93

Need this to be fixed ?

E.g. for Mindelo it is named:
sups_oscm_mwr00_l2_prw_v00_20210916000000.....

HolgerPollyNet, 2022-04-12:
This should work in line 93, right?: 
_sprintf('*_mwr00_l2_prw_v00_%s*.nc', ..._

HolgerPollyNet, 2022-04-13:
The folder of course I changed, but the filename structure to search for was too much hardcoded, as shown above, so with the proposed change it workded, an MWR files were found - But the calibration is not yet working, something I have to dive in together with @ulysses78

HolgerPollyNet, 2023-02-07:
any progress on this @ulysses78  ?

ulysses78, 2023-02-07:
Not yet.

ulysses78, 2023-07-12:
I added a function called ```lib/io/readMWR_cloudnet.m``` with the option to use IWV cloudnet products from HATPRO-MWR devices. This function reads the cloudnet product file, which is already stored on rsd2-server. This function of course is recognized within the file ```lib/io/readIWV.m``` as a new option, besides 'AERONET' and 'MWR'. Whereas 'MWR' means MWR_pro products.
To use this new function, one has to change the following key/values in the polly_global_config.json file:

"IWV_instrument": "MWR_cloudnet",
"MWRFolder": "/data/level1b/cloudnetpy/products",

HolgerPollyNet, 2024-07-26:
Is solved and can be closed, can't it? @ulysses78 @martin-rdz

closed 2024-07-26 by HolgerPollyNet