PollyNET/Pollynet_Processing_Chain #160: Raman bsc near-range height-shifted

HolgerPollyNet, 2022-06-07:
https://polly.tropos.de/datavis/location/35/25/?dates=[2019-12-02T00:00:00,2019-12-03T00:00:00]

[image]

Any ideas where to search? Test case 2019-12-02 arielle, config: arielle_config_mosaic.

HolgerPollyNet, 2022-06-07:
currently running without dt correction - just for checking

ZPYin, 2022-06-08:
It was caused by wrong indexing of far-range Raman channel for near-range Raman retrieval. I have fixed that in a new commit and make it updated in the server. 

You can check the re-processed results via the link below

https://polly.tropos.de/datavis/location/35/25/?dates=%5B2019-12-02T00:00:00,2019-12-03T00:00:00%5D

I would close this issue at the moment. If you have new comments in terms of this issue, feel free to re-open it.

HolgerPollyNet, 2022-06-08:
Great! Many thanks!

HolgerPollyNet, 2022-06-08:
I have one more comment: Do you have any clue why the near-range raman bsc can be comnpletely negatoive between 0.5 and 3 km, while the ref height is between 2.5. and 3 km? In my opinion this should not be possible: 
[image]

@ZPYin @griesche

HolgerPollyNet, 2022-06-08:
sorry, my mistake

closed 2022-06-08 by HolgerPollyNet