PollyNET/Pollynet_Processing_Chain #162: Documentation of automatic. Ref height search

HolgerPollyNet, 2022-06-15:
The ACTRIS community is interested in using the Rayleigh fitting procedure as done by us for the SCC. Can we make a brief documentation, stating what input parameter are used, which algorithms are used and especially which thresholds apply. In principle a flow chart or so would be perfect. But to know, on which parameters it is decided if a certain ref height is seen as appropriate or not.   We could use this also then to extent our online documentation. Their plan is to use python for the new developments, but I see benefit for all of us.

@ZPYin would you be willing to make  a start?
@ulysses78 could you, if once finished digest it to the online documentation?

ZPYin, 2022-06-15:
OK, I will prepare some slides to demonstrate the workflow and some theoretical background for supporting this. 
But it may take a few days, due to some time-consuming tasks at my hand. I will make it ready before next Thuesday, just let you know.

ulysses78, 2022-06-15:
Sure, I will take care of the online documentation.

ZPYin, 2022-06-21:
[Pollynet_Processing_Chain_Rayleigh_Fit.pptx](https://github@HolgerPollyNet com/PollyNET/Pollynet_Processing_Chain/files/8946829/Pollynet_Processing_Chain_Rayleigh_Fit.pptx)

@HolgerPollyNet @ulysses78 Please see the attached file. It shows some descriptions of Rayleigh Fit algorithm used in Picasso.  

I didn't have strict and consice mathmatical expressions for this algorithm. And some improvements that I did were also based on my experience on data analysis of long-term dataset from different pollynet campaigns. I can't provide all these details, since I lost some of the evidence after code development during these three years. Nevertheless, the main ideas are included.
 
If something is unclear, just put your comments below. I will come back to this thread when it's needed.

HolgerPollyNet, 2022-06-23:
thanks, very helpful, can be closes currently.

closed 2022-06-23 by HolgerPollyNet