PollyNET/Pollynet_Processing_Chain #168: Restructuring the visualization of long-term  calibration results

HolgerPollyNet, 2022-07-11:
Currently, the calibration results (Lidar constant, depo, water vapor) are calculated, stored in a DB and finally plotted in one plot.

This plot is very, very helpful to check the system behavioiur, however with more and more parameter it gets overloaded.

In my opinion, the meanwhile very heavy loaded https://github.com/PollyNET/Pollynet_Processing_Chain/blob/8a0c6e9d156cf99abf3984992abd179a26a64b22/lib/visualization/pollyxt_ift_displayLTLCali.py needs to be restructured.

In my opinion, two plots wopuld be the best. 
One for lidar constants: 355, 355_NF, 532, 532_NF, 1064.....and one for depol and water vapor constant would be good for a more clear visibilty and future developments.

ulysses78, 2022-07-12:
I started to seperate the plots. As an example I took the measurements from pollyxt_cpv.
Please have a look and tell me what you think...

[image]

[image]

HolgerPollyNet, 2024-07-26:
@ulysses78 can be closed, correct?

closed 2024-07-26 by HolgerPollyNet