PollyNET/Pollynet_Processing_Chain #175: bug on dev-branch when analyzing the meteorological data

ulysses78, 2022-10-06:
@ZPYin , Thanks for your fast support at this point 0954aa99bb3ab002a42bfd710f01996dbcf4243d.
But now there is a new bug when running picasso on the dev-branch. This seems to have something to do with the used interpolation method, when analyzing the meteorological data. Do you have any clue?

Here is the bug-report:

```
[2022-10-06 06:46:22] Start loading meteorological data.
Error using griddedInterpolant
The grid vectors are not strictly monotonic increasing.

Error in interp1 (line 183)
        F = griddedInterpolant(X,V,method);

Error in interpMeteor (line 37)
yOut = interp1(x(validIndx), y(validIndx), xOut, method, 'extrap');

Error in loadMeteor (line 117)
    tempI = interpMeteor(altRaw, tempRaw, asl);

Error in picassoProcV3 (line 749)
[temp, pres, relh, ~, ~, meteorAttri] = loadMeteor(clFreGrpTimes, data.alt, ...

Error in picassoProcTodolist (line 62)
    reportTmp = picassoProcV3(pollyDataFile, pollyDataTasks.pollyType{iTask},
    ...

>> >> >> Log4perl: Seems like no initialization happened. Forgot to call init()?
[2022/10/06 06:46:25] [INFO] [B174.Logging] [Log4perl initialised with /pollyhome/Bildermacher2/pollyAPP/config/log.conf.private] [B174::Logging::BEGIN 57]
Create pidfile and process data
Check max. number of lines in done_filelist
[2022/10/06 06:46:25] [INFO] [B174.Model.Role.TimeMeasurementRole] ['BUILD' was started in 'B174::Model::Util::ProcessDoneFileList' with time measurement.] [Class::MOP::Class:::around 89]
[2022/10/06 06:46:25] [DEBUG] [parserupdater] [Begin: parse done file and update lidar product.] [B174::Model::Util::ProcessDoneFileList::BUILD 111]
[2022/10/06 06:46:25] [INFO] [parserupdater] [Filename: /pollyhome/Bildermacher2/done_filelist/done_filelist_andi.txt] [B174::Model::Util::ProcessDoneFileList::BUILD 112]
[2022/10/06 06:46:25] [DEBUG] [parserupdater] [Begin: parse file] [B174::Model::Util::ProcessDoneFileList::BUILD 127]                                                         [2022/10/06 06:46:25] [INFO] [B174.Model.Role.TimeMeasurementRole] ['BUILD' was started in 'B174::Model::Util::ProcessDoneFileList::ReadDoneFileList' with time measurement.]
[Class::MOP::Class:::around 89]
[2022/10/06 06:46:25] [DEBUG] [parserupdater] [Start: read file /pollyhome/Bildermacher2/done_filelist/done_filelist_andi.txt] [B174::Model::Util::ProcessDoneFileList::ReadDoneFileList::BUILD 99]
[2022/10/06 06:46:25] [DEBUG] [parserupdater] [End: read file /pollyhome/Bildermacher2/done_filelist/done_filelist_andi.txt] [B174::Model::Util::ProcessDoneFileList::ReadDoneFileList::BUILD 103]
[2022/10/06 06:46:25] [INFO] [parserupdater] [Read done_filelist finished. 0 records are parsed.] [B174::Model::Util::ProcessDoneFileList::ReadDoneFileList::assign_data 238]
[2022/10/06 06:46:25] [INFO] [B174.Model.Role.TimeMeasurementRole] [Execution time of 'BUILD' in 'B174::Model::Util::ProcessDoneFileList::ReadDoneFileList' was 0.003454 seconds.] [Class::MOP::Class:::around 102]
Attribute (total_number_of_valid_invalid_records) does not pass the type constraint because: Validation failed for 'Int' with value undef at accessor B174::Model::Util::ProcessDoneFileList::total_number_of_valid_invalid_records (defined at /pollyhome/Bildermacher2/pollyAPP/src/util/../model/B174/Model/Util/ProcessDoneFileList.pm line 95) line 4
        B174::Model::Util::ProcessDoneFileList::total_number_of_valid_invalid_records('B174::Model::Util::ProcessDoneFileList=HASH(0x6b8b668)', undef) called at /pollyhome/Bildermacher2/pollyAPP/src/util/../model/B174/Model/Util/ProcessDoneFileList.pm line 130
        B174::Model::Util::ProcessDoneFileList::BUILD('B174::Model::Util::ProcessDoneFileList=HASH(0x6b8b668)', 'HASH(0x6c019d8)') called at /pollyhome/Bildermacher2/pollyAPP/src/util/../model/B174/Model/Role/TimeMeasurementRole.pm line 95
        Class::MOP::Class:::around('CODE(0x5cd67c0)', 'B174::Model::Util::ProcessDoneFileList=HASH(0x6b8b668)', 'HASH(0x6c019d8)') called at /pollyhome/Bildermacher2/.perlbrew/libs/perl-5.22.2@devel/lib/perl5/x86_64-linux/Class/MOP/Method/Wrapped.pm line 164
        B174::Model::Util::ProcessDoneFileList::_wrapped_BUILD('B174::Model::Util::ProcessDoneFileList=HASH(0x6b8b668)', 'HASH(0x6c019d8)') called at /pollyhome/Bildermacher2/.perlbrew/libs/perl-5.22.2@devel/lib/perl5/x86_64-linux/Class/MOP/Method/Wrapped.pm line 95
        B174::Model::Util::ProcessDoneFileList::BUILD('B174::Model::Util::ProcessDoneFileList=HASH(0x6b8b668)', 'HASH(0x6c019d8)') called at /pollyhome/Bildermacher2/.perlbrew/libs/perl-5.22.2@devel/lib/perl5/x86_64-linux/Class/MOP/Method.pm line 128
        Class::MOP::Method::execute('Class::MOP::Method::Wrapped=HASH(0x3596f60)', 'B174::Model::Util::ProcessDoneFileList=HASH(0x6b8b668)', 'HASH(0x6c019d8)') called at /pollyhome/Bildermacher2/.perlbrew/libs/perl-5.22.2@devel/lib/perl5/x86_64-linux/Moose/Object.pm line 54
        Moose::Object::BUILDALL('B174::Model::Util::ProcessDoneFileList=HASH(0x6b8b668)', 'HASH(0x6c019d8)') called at /pollyhome/Bildermacher2/.perlbrew/libs/perl-5.22.2@devel/lib/perl5/x86_64-linux/Moose/Meta/Class.pm line 279
        Moose::Meta::Class::new_object('Moose::Meta::Class=HASH(0x3f64580)', 'HASH(0x6c019d8)') called at /pollyhome/Bildermacher2/.perlbrew/libs/perl-5.22.2@devel/lib/perl5/x86_64-linux/Moose/Object.pm line 24
        Moose::Object::new('B174::Model::Util::ProcessDoneFileList', 'filename', '/pollyhome/Bildermacher2/done_filelist/done_filelist_andi.txt', 'is_test', 0, 'check_image_exists', 1) called at /pollyhome/Bildermacher2/pollyAPP/src/util/add_new_data2pollydb.pl line 153
        main::main at /pollyhome/Bildermacher2/pollyAPP/src/util/add_new_data2pollydb.pl line 177
export PERL5LIB=/pollyhome/Bildermacher2/.perlbrew/libs/perl-5.22.2@devel/lib/perl5;/pollyhome/Bildermacher2/perl5/perlbrew/perls/perl-5.22.2/bin/perl /pollyhome/Bildermacher2/pollyAPP/src/util/add_new_data2pollydb.pl /pollyhome/Bildermacher2/done_filelist/done_filelist_andi.txt;: Signal 127
>> Finish

```

ZPYin, 2022-10-06:
This issue was associated with GDAS1 profiles, not the fault of Picasso (in my opinion). I have discussed it Hogler (@HolgerPollyNet ) before, please see issue #135 for more details.

Below is the screenshot of the latest GDAS1 profile for Evora (evora_20221005_06_39.0_-8.0.gdas1)

[image]

closed 2022-10-06 by ZPYin

ulysses78, 2022-10-06:
@ZPYin ... many thanks for clearification. I started a new picasso run. Now the GDAS1 data seems to be ok and Picasso is doing fine :-)