PollyNET/Pollynet_Processing_Chain #176: use standard atmosphere values for meteorological data if GDAS values are inconsistent

ulysses78, 2022-10-11:
When picasso is running, sometimes the following error occurs:

```
[2022-10-11 09:15:28] Start loading meteorological data.
{Error using griddedInterpolant
The grid vectors are not strictly monotonic increasing.
 
Error in interp1 (line 183)
         F = griddedInterpolant(X,V,method);
 
Error in interpMeteor (line 37)
yOut = interp1(x(validIndx), y(validIndx), xOut, method, 'extrap');
 
Error in loadMeteor (line 117)
     tempI = interpMeteor(altRaw, tempRaw, asl);
 
 
Error in picassoProcV3 (line 749)
[temp, pres, relh, ~, ~, meteorAttri] = loadMeteor(clFreGrpTimes, data.alt, ...
 
Error in picassoProcTodolist (line 62)
     reportTmp = picassoProcV3(pollyDataFile, pollyDataTasks.pollyType{iTask},
     ...
}
>> >> >> Log4perl: Seems like no initialization happened. Forgot to call init()?
[2022/10/11 09:15:30] [INFO] [B174.Logging] [Log4perl initialised with /pollyhome/Bildermacher2/pollyAPP/config/log.conf.private]
[B174::Logging::BEGIN 57]
Create pidfile and process data
Check max. number of lines in done_filelist
sh: /pollyhome/Bildermacher2/done_filelist/done_filelist.txt: No such file or directory wc failed: 256 at /pollyhome/Bildermacher2/pollyAPP/src/util/add_new_data2pollydb.pl line 138.
>> Finish

```

This is caused due to some inconsistent GDAS1 data as can be seen in the following GDAS1-file from Leipzig (/data/level1a/model/gdas1/profiles/leipzig/2022/10/leipzig_20221011_00_51.3_12.4.gdas1):

```
Profile Time:  22 10 11  0  0
 Used Nearest Grid Point (  13, 142) to Lat:    51.30, Lon:    12.40
        2D Fields
        PRSS  MSLP  TPP6  UMOF  VMOF  SHTF  DSWF  RH2M  U10M  V10M  T02M  TCLD  SHGT  CAPE  CINH  LISD  LIB4  PBLH  TMPS  CPPA  SOLM  CSNO  CICE  CFZR  CRAI  LHTF  LCLD  MCLD  HCLD  WDIR  WSPD
         hPa   hPa    mm              W/m2  W/m2     %   m/s   m/s    oC     %     m                             m    oC                                      W/m2     %     %     %   deg   m/s
  1013     0     0     0     0     0     0     0     0     0     0     0     0     0     0     0     0     0     0     0     0     0     0     0     0     0     0     0     0     0
        3D Fields
        HGTS  TEMP  UWND  VWND  WWND  RELH     TPOT  WDIR  WSPD
           m    oC   m/s   m/s  mb/h     %       oK   deg   m/s
  1000     0     0     0     0     0     0    273.2   NaN   0.0
   975     0     0     0     0     0     0    275.1   NaN   0.0
   950     0     0     0     0     0     0    277.2   NaN   0.0
   925     0     0     0     0     0     0    279.3   NaN   0.0
   900     0     0     0     0     0     0    281.5   NaN   0.0
   850     0     0     0     0     0     0    286.2   NaN   0.0
   800     0     0     0     0     0     0    291.2   NaN   0.0
   750     0     0     0     0     0     0    296.6   NaN   0.0
   700     0     0     0     0     0     0    302.5   NaN   0.0
   650  3675  -4.8  15.4  -2.3   4.0  16.0    303.6 278.6  15.6
   600  4301  -7.8  13.3  -5.6   2.6   5.4    307.1 292.7  14.5
   550  4970 -13.2  13.0  -5.7   4.4   4.3    308.5 293.6  14.1
   500  5687 -19.1  14.2  -4.5   1.8   3.8    309.8 287.6  14.9
   450  6460 -26.0  15.3  -3.4     0   7.3    310.5 282.6  15.6
   400  7301 -32.3 -2966     0     0     0    313.0  90.02965.6
   350     0     0     0     0     0     0    368.8  90.02965.6
   300     0     0     0     0     0     0    385.4  90.02965.6
   250     0     0     0     0     0     0    406.1  90.02965.6
   200     0     0     0     0     0     0    432.8  90.02965.6
   150     0     0     0     0     0     0    470.0  90.02965.6
   100     0     0     0     0     0     0    527.7  90.02965.6
    50     0     0     0     0                643.4  90.02965.6
    20     0     0     0     0                836.2  90.02965.6
```

This kind of error was mentioned before (https://github.com/PollyNET/Pollynet_Processing_Chain/issues/135#issuecomment-1007895581) and was somehow solved if the values for height and temperature equal "0". If so, the standard atmosphere values have to be choosen.

Now we have to find a way to detect if GDAS1 values are linear in terms of height-values (e.g.: if the slope in the height values is always positive). If so, the GDAS1 values can be taken into account. If not, the standard atmosphere values have to be choosen.

@HolgerPollyNet ... do you agree with this approach?

HolgerPollyNet, 2022-10-11:
sounds reasonable! Go for it!

HolgerPollyNet, 2022-10-11:
or you just implement the try and catch function in maltab..dirty but could work ;-)

ZPYin, 2022-10-17:
Let's see if this issue can be closed...

ulysses78, 2022-10-18:
I pulled the latest commit (2e7310a556ea11326b970bcc5d3a7400e75ba655) from dev-branch and let it run. so far it seems to work. I will close this issue for now.

closed 2022-10-18 by ulysses78