PollyNET/Pollynet_Processing_Chain #184: Store SNR from all channels as  netcdf

HolgerPollyNet, 2022-12-09:
or is this already possible with the safe product list option @ZPYin ?

ZPYin, 2022-12-09:
It's not in the Product Save List yet...

See the complete product list below:
|short name|description|
|:--:|:---|
|`overlap`|overlap functions|
|`aerProfFR`|Profiles of aerosol optical properties derived from far-range channels|
|`aerProfNR`|Profiles of aerosol optical properties derived from near-range channels|
|`aerProfOC`|Profiles of aerosol optical properties derived from overlap-corrected siganls|
|`aerAttBetaFR`|Attenuated backscatter from far-range channels|
|`aerAttBetaNR`|Attenuated backscatter from near-range channels|
|`aerAttBetaOC`|Attenuated backscatter from overlap-corrected signals|
|`WVMR_RH`|water vapor mixing ratio and relative humidity|
|`volDepol`|volume depolarization ratio with high spatiotemporal resolution|
|`quasiV1`|High spatiotemporal resolved optical properties derived from quasi-retrieval (version 1)|
|`quasiV2`|High spatiotemporal resolved optical properties derived from quasi-retrieval (version 2)|
|`tc`|target classifications derived from quasiV1|
|`tcv2`|target classifications derived from quasiV2|
|`cloudinfo`|cloud mask|

You can create new entry of product list in [`polly_global_config,json`](https://github.com/PollyNET/Pollynet_Processing_Chain/blob/master/lib/config/polly_global_config.json), and define operations for the new entry in `PicassoProV3`
https://github.com/PollyNET/Pollynet_Processing_Chain/blob/8a0c6e9d156cf99abf3984992abd179a26a64b22/lib/interface/picassoProcV3.m#L4713-L4800

HolgerPollyNet, 2023-02-07:
After thinking about, I do not know if this is really needed. --> but maybe for higher versions