PollyNET/Pollynet_Processing_Chain #188: bug when "flagDeleteData": true

HolgerPollyNet, 2023-01-03:
If 

    "flagDeleteData": true,

then the data file is deleted before it is read.....

ZPYin, 2023-01-04:
I guess the error was caused by the line below (please correct me if not):
https://github.com/PollyNET/Pollynet_Processing_Chain/blob/1faa5a7ec46dc32c3613abaeb71f47074e9c7859/lib/io/readPollyRawData.m#L158

This line will throw error of "Could not open file for reading" after the netcdf file was deleted before in line 121-123 (so when `flagDeleteData`: true)
https://github.com/PollyNET/Pollynet_Processing_Chain/blob/1faa5a7ec46dc32c3613abaeb71f47074e9c7859/lib/io/readPollyRawData.m#L121-L123

I recommend to delete line 158 to resolve this issue, because `measurement_time` was already loaded before (see line 101). It's not necessary here. (Please confirm @ulysses78 )
https://github.com/PollyNET/Pollynet_Processing_Chain/blob/1faa5a7ec46dc32c3613abaeb71f47074e9c7859/lib/io/readPollyRawData.m#L158
https://github.com/PollyNET/Pollynet_Processing_Chain/blob/1faa5a7ec46dc32c3613abaeb71f47074e9c7859/lib/io/readPollyRawData.m#L101

ZPYin, 2023-01-05:
> As @ulysses78 is still out of office I made a temporary bug fix with commit 'how to link the commit??' [1cea589](https://github.com/PollyNET/Pollynet_Processing_Chain/commit/1cea589f16763bd1f03c883d0cef91eb3a33f194)

The usual way that I did for linking the issue, is to add additional line in the committing message with the issue number, for instance:

(git bash)
```bash
git commit -m "bugfixes: associated with #188"
```

Then the commit number will be shown in the issue #188 after pushing to GitHub repo.

closed 2023-01-16 by HolgerPollyNet