PollyNET/Pollynet_Processing_Chain #193: handling of corrupted /small files

HolgerPollyNet, 2023-01-24:
error in case of small files with cloud screening - see below

For example: 2020_02_01_Sat_ARI_18_12_34.nc (less than 10 min). How can one prevent the processing chain to abort? Better would be to discard the file....

error message:
  In picassoProcTodolist at 62
  In picassoProcHistoryData at 69
[2023-01-24 09:04:53] Finish.
[2023-01-24 09:04:53] Start cloud screening.
Error using cloudDetect_Zhao (line 68)
Dimensions are not matched!

Error in cloudScreen (line 90)
    [flagCloudFree, layerStatus] = cloudDetect_Zhao(time, height, signal, ...

Error in picassoProcV3 (line 484)
    [flagCloudFree_FR, cloudMask] = cloudScreen(data.mTime, data.height, ...

Error in picassoProcTodolist (line 62)
    reportTmp = picassoProcV3(pollyDataFile, pollyDataTasks.pollyType{iTask},
    ...

Error in picassoProcHistoryData (line 69)
picassoProcTodolist(PicassoConfigFile);

ZPYin, 2023-01-27:
This was caused by typos of using channel flags. It has been fixed in the commit [2eb1eas](https://github.com/PollyNET/Pollynet_Processing_Chain/commit/2eb1eab89f979afa1c64d431d4b07c951b286f90) (see above).

HolgerPollyNet, 2023-02-07:
So the issue can be closed?

closed 2023-03-06 by HolgerPollyNet