PollyNET/Pollynet_Processing_Chain #202: corrupted time in netcdf --> chain stops

HolgerPollyNet, 2023-02-28:
In case of power failures etc. it can happen that the netcdf is kind of corrupted in the last entry:
[image]
In the example here, the time values are completely nonsense. But currently the processing chain is aborting while plotting the long-term cali. Is there a way to avoid the stopping of the chain but to give a warning?

HolgerPollyNet, 2023-02-28:
2023-02-28 00:14:56] --> start diplaying lidar housekeeping data.
{Error using timelabellayout (line 166)
The span is too large.

Error in pollyxt_displayHousekeeping (line 21)
[xtick, xtickstr] = timelabellayout(data.mTime, 'HH:MM');

Error in pollyDisplayHousekeeping (line 19)
    pollyxt_displayHousekeeping(data);

Error in picassoProcV3 (line 4844)
    pollyDisplayHousekeeping(data);

Error in picassoProcTodolist (line 62)
    reportTmp = picassoProcV3(pollyDataFile, pollyDataTasks.pollyType{iTask},

ZPYin, 2023-02-28:
To correct it a little bit... The error happens in the plotting routine of housekeeping data, while it detects an unreasonably large time interval between the first and the last profile. 

To keep processing chain running without collapse, we can can a `try..catch...` in `pollyxt_displayHousekeeping.m`. which I will implement in the next commit. 

But this seems not perfect for this situation. What about inserting another condition check in the preprocessing module, to rule out profiles with strange timestamp (let's say negative values...)?

HolgerPollyNet, 2023-02-28:
I completely agree. Try and catch can be used as workaround, but more sustainable is the preproc solution!

ZPYin, 2023-02-28:
see below vol. depol. for PollyXT_CPV with wrong timestamp

[image]

HolgerPollyNet, 2023-02-28:
@ulysses78 can you pull the new version to the server?

ulysses78, 2023-02-28:
job done!

ulysses78, 2023-03-06:
this one can be closed, right?

commit 75a8c9e1 (2023-02-28): Enhancement: exclude wrong profiles

Excluding profiles with wrong (negative) timestamp. And add `try-catch` block inside data visualization function to continue processing when error happens.

Close #202
closed 2023-03-06 by HolgerPollyNet