PollyNET/Pollynet_Processing_Chain #214: Change plotting mode of time-height plots to 24 h

HolgerPollyNet, 2023-07-07:
@ulysses78 , can you briefly discuss what you are currently working on, so that everybody is informed and link to the respective branch?

Thanks

ulysses78, 2023-07-14:
### aim of this project
This project combines two objectives:
1. The first goal of this project is to have 24h time-height plots for e.g. attbsc, voldpol, categorization, ... and so on. Until now those plots are splitted into 4x6h plots (in most cases). This is due to the measurement process. With the 24h-plots the diagrams can be interpreted better and more easily. Such 24h plots are also used by some other cooperation partners, e.g. cloudnet (fmi). Profile plots are not part of this project.
2. The second goal is to speed up the python plotting procedure

### picassos plotting procedure: status quo
Until now Picasso generates the plots not directly from the level1-nc-files, but from the processed results using matlab. Those results are temporarily stored in a .mat file. The content of this .mat file is then used for plotting. The plots are generated by calling python-scripts within matlab. The python plotting procedure at the moment uses the `pcolormesh` function. Advantage of this function is the nice handling of measurement gaps, but the big disadvantage is the slow processing speed. This is a critical major issue when processing and plotting huge amounts of polly data.

### the new approach - 24h plotting
In the new approach one has to combine all the informations from one day. This is done by merging the level0 files first. The level0 files are only merged if the polly settings haven't changed during the day.
Afterwards the merged file is proccessed by using picasso until the results are stored in the level1-files.
The new plotting scheme uses the information stored in the level1-files. It uses the `imshow` python-function. The advanatage of this function is its fast processing speed (approx. 8x faster compared to `pcolormesh`). The disadavntage is the bad handling of measurement gaps. To avoid this disadvantage I used matrixfilling of the data blocks. In other words: for one day an empty matrix is created and is filled with the available measurement data information (e.g. attbsc, voldepol, ...).

### usage
At the moment this 24h plotting is in experimental state. It runs in a seperated picasso folder on the rsd2 server under `/pollyhome/Bildermacher2/experimental/Pollynet_Processing_Chain`. 
One has to add the new key `flagEnableDataVisualization24h` to the picasso-config-file and set this one to `true`.
When picasso reaches the visualization step for the 24h plots it calls the following python script: `/pollyhome/Bildermacher2/src/pypolly/pypolly_display_all.py`.
To generate 24h plots for a specified location+date one has to run the following bash script: `/pollyhome/Bildermacher2/src/concat_pollyxt_lvl0/picasso_go24.sh`.
The syntax is as follows:
`/pollyhome/Bildermacher2/src/concat_pollyxt_lvl0/picasso_go24.sh  --startdate YYYYMMDD --enddate YYYYMMDD -d "$DEVICE" -c $PICASSO_CONFIG_FILE --fproc`

In the following chart you can see the workflow of the `picasso_go24.sh` script:

[image]

ulysses78, 2023-07-14:
### First test quicklooks
To view the actual running test quicklooks, Rico Hengst set up a local web server@rsd2 running at `localhost:3333`.
To access this test instance of pollyAPP use portforwarding and create a ssh connection to Bildermacher2@rsd2
`ssh -L 12349:localhost:3333 Bildermacher2@rsd2`
If the connection is established, please use your local webbrowser `http://localhost:12349/` and connect localhost at the named port. This test instance is a little bit slow, so please be patient.

ulysses78, 2023-07-14:
FYI: @ZPYin @rico-hengst @cristoferjimenez @AthenaAugousta @gast-ben @martin-rdz @Ronny-TROPOS

HolgerPollyNet, 2023-07-14:
should this really be public?

elmarinou, 2023-07-14:
Ioanna has a code to make tailored pocasso plots, where the user can choose time start, time end, height range, and colorbar max values for volume depol. and att. bp. 
Do you think we should consider this option as a new tailored plots topic? or can this code you are developing now also have an option like this? 
Have a nice weekend!

ZPYin, 2023-07-14:
> FYI: @ZPYin @rico-hengst @cristoferjimenez @AthenaAugousta @gast-ben @martin-rdz @Ronny-TROPOS

Hi Andi(@ulysses78)! Thanks for your efforts to make it work.

From my point of view, there are two different things in between. 
- PollyNET Processing Chain
- pollyAPP

PollyNET Processing Chain is an **public** repository to show our partners how to deal with PollyXT data. And pollyAPP is so far an **internal** application for showing pollynet-related content. This repository only focuses PollyNET Processing Chain staffs.

I will hide your posts in terms of pollyAPP. But free feel to communicate pollyAPP content through emails.

ulysses78, 2023-07-14:
I just mentioned this special version of the pollyapp, because there you can see the 24h plots for pollyxt_lacros and pollyxt_cpv.

ulysses78, 2023-07-17:
For security reasons I shifted the detailed description of this project to our internal doku-wiki pages (only accessible via vpn):
https://rsdwiki.tropos.de/doku.php?id=data_processing:picasso:picasso24h

ulysses78, 2023-07-17:
> Ioanna has a code to make tailored pocasso plots, where the user can choose time start, time end, height range, and colorbar max values for volume depol. and att. bp. Do you think we should consider [...]

We also have a pypi-project called `pylarda` (https://github.com/lacros-tropos/larda). With this one can generate time-height plots for different devices (e.g.: pollyxt, hatpro, halo, mira35, rpg94, ...). But with pylarda it is not possible to plot quicklooks during the picasso-processing.
The aim of this picasso24h-plot project is to generate 24h-quicklooks during the picasso is processing the (new) measurement data and speed up the plotting process.

HolgerPollyNet, 2023-08-07:
> Hi Andi(@ulysses78)! Thanks for your efforts to make it work.
>
> From my point of view, there are two different things in between.
>
>     * PollyNET Processing Chain
>
>     * pollyAPP
>
>
> PollyNET [...]

Actually, the 24h capability is kind of a new fork from the PPC. So its fine to leave it here... Best

HolgerPollyNet, 2024-07-26:
Was successfully implemented! with the release of V4

closed 2024-07-26 by HolgerPollyNet