PollyNET/Pollynet_Processing_Chain #224: Poliphon params to config file

HolgerPollyNet, 2023-09-01:
In general, but maybe for a future version it would be also great to source out everything that could be adjusted to the config file.

In your case it would be:
Dd = [0.23, 0.31, 0.25];
Dnd = [0.05, 0.05, 0.05];

and then, the used paramters should be stored in the netcdf file

AthenaAugousta, 2023-11-20:
Note to self: 
In the NetCDF file the variable 'uncertainty_aerBsc1064__nd1' needs to be renamed to 'uncertainty_aerBsc1064_klett_nd1'

AthenaAugousta, 2024-05-21:
> Note to self: In the NetCDF file the variable 'uncertainty_aerBsc1064__nd1' needs to be renamed to 'uncertainty_aerBsc1064_klett_nd1'

--> Done

HolgerPollyNet, 2024-05-22:
is the general issue still open? concerning the used parameters? @AthenaAugousta

AthenaAugousta, 2024-05-22:
> is the general issue still open? concerning the used parameters? @AthenaAugousta

Yes, will work on that soon though!