PollyNET/Pollynet_Processing_Chain #229: fetch + ml22

HolgerPollyNet, 2023-10-20:
When running PPC with Matlab22, I still get the following message:

Start to fetch polly data filenames for  from polly_202009
{Undefined function 'fetch' for input arguments of type 'struct'.

Error in writeTodoListAuto (line 130)
    dbRet = fetch(res);
} 

Any idea hwat to do @ZPYin

ZPYin, 2023-10-21:
> This is a issue that can only occure for running in the server. So without public interest!

I have checked this issue in the server and I suspect this may associate with two factors.

**Java Database Driver (JDBC) for MySQL**

I found out that Matlab cannot connect to mysql database as no suitable JDBC can be found (see below).
[image]

There is different JDBCs for different Database. I have put a general JDBC for sqlite in the PPC repository, but didn't do that for MySQL. You can try to install MySQL JDBC through the link below
https://de.mathworks.com/products/database/driver-installation.html

But this makes me surprized as PPC runs quite well, as I see the update of PollyNET webpage continuously. If MySQL cannot be connected, this would not happen.

**Empty line in PPC config link file**
filename: `pollynet_processing_chain_config_links_rsd2.xlsx`
line: 150
[image]

This may trigger error when parsing data from Polly MySQL database, as it suggests a polly system with an empty label. You can try to remove this empty line and re-run PPC to see if this error still exists.

Hope it helps. If not, please update the situation, in case I can have more information to dive into.

ZPYin, 2023-10-23:
I have added the JDBC for mysql into the matlab search path. And it works now. 

The empty line was also removed, but it seems not to be related with this issue though.

Now feel free to close this issue.

HolgerPollyNet, 2023-10-23:
Thanks a lot!!!! It works!

closed 2023-10-23 by HolgerPollyNet