PollyNET/Pollynet_Processing_Chain #235: store configuration file in netcdf

HolgerPollyNet, 2023-12-14:
To allow reproducibility, I would propose to store the whole config file which has been used, i.e. all parameters, in the netcdf as well. Maybe as simple character variable first. But this would help to identify issues in case the processing was made elsewhere....

cristoferjimenez, 2024-07-31:
Hi Holger,

I think in that case, one needs to adapt the types of each parameter. I tried this change and it works.

Best regards, Cristofer

s = strings
for i=1:size(names)
    
    type=   class(values{i});
    if strcmp(type,'cell')
        values2=num2str(str2double(values{i})) ;
    elseif strcmp(type,'logical')
        values2=num2str(double(values{i}));
    else
        values2=(values{i});
        
        if strcmp(type,'double')
            
            if size(values2,1)>1
                values2=num2str(reshape(values2,1,size(values2,1)*size(values2,2)));
            else
                values2=num2str(values2);
            end
        end
    end
    
    s = strcat(s,names{i},',',string(values2),',')
end

ZPYin, 2024-08-01:
Hi Holger,

If you want to convert a Matlab struct to a char array or a string, you need to specify the output format for different data types (as suggested by @cristoferjimenez ). 

I modified the code (from @cristoferjimenez ) a little bit to make it support the conversion of `PollyConfig` struct. Please see the two little functions in the link below.

[matlab-funcs-struct-2-char.zip](https://github.com/user-attachments/files/16451310/matlab-funcs-struct-2-char.zip)

Below is an output test.

PollyConfig
```json
{
    "flagCorrectFalseMShots": true,	
    "flagFilterFalseMShots": false,
    "flagDTCor": true,

    "flagWVCalibration": true,
    "flagUsePreviousDepolCali": true,
    "flagUsePreviousWVconst": true,
    "flagUsePreviousLC": true,

    "dataFileFormat": "(?<year>\\d{4})_(?<month>\\d{2})_(?<day>\\d{2})_\\w*_(?<hour>\\d{2})_(?<minute>\\d{2})_(?<second>\\d{2})\\w*.nc",
    "gdas1Site": "cabauw",
    "AERONETSite": "Cabauw",

    "max_height_bin": 3000,
    "first_range_gate_indx": [261, 261, 261, 261, 261, 261, 261, 261, 262, 262],
    "first_range_gate_height": 78.75,
    "dtCorMode": 2,
    "dt": [[0.0, 0.972992, 0.00353332, -7.90981e-006, 1.06451e-007, 1.42895e-009], 
        [0, 1.0117, -0.0014, 0.0002, -0.0000, 0.0000],
        [0, 0.9674, 0.0023, 0.0000, 0.0000, 0.0000],
        [0, 0.9929, 0.0000, 0.0001, -0.0000, 0.0000],
        [0, 0.9843, 0.0022, 0.0001, -0.0000, 0.0000],
        [0, 0.9391, 0.0063, -0.0001, 0.0000, -0.0000],
        [0, 1.0035, 0.0003, 0.0001, -0.0000, 0.0000],
        [0, 1.0000, 0, 0, 0, 0],
        [0, 1.0000, 0.0029, 0.0000, 0.0000, 0.0000],
        [0, 1.0000, 0.0028, 0.0000, 0.0000, 0.0000]],
    "bgCorRangeIndx": [10, 240],
    "mask_SNRmin": [1, 1, 1, 0.01, 1, 1, 1, 1, 1, 1], 
    "init_depAng": 19.99,
    "maskDepCalAng": ["none", "none", "p", "p", "p", "p", "p", "p", "p", "p", "none", "none", "n", "n", "n", "n", "n", "n", "n", "n", "none"],
    "depol_cal_minbin_532": 40,
    "depol_cal_maxbin_532": 300,
    "depol_cal_SNRmin_532": [1, 1, 1, 1],
    "depol_cal_sigMax_532": [1500, 1500, 1500, 1500],
    "rel_std_dplus_532": 0.2,
    "rel_std_dminus_532": 0.2,
    "depol_cal_segmentLen_532": 40,
    "depol_cal_smoothWin_532": 8,
    "depol_cal_minbin_355": 40,
    "depol_cal_maxbin_355": 300,
    "depol_cal_SNRmin_355": [2, 2, 2, 2],
    "depol_cal_sigMax_355": [1000, 1000, 1000, 1000],
    "rel_std_dplus_355": 0.2,
    "rel_std_dminus_355": 0.2,
    "depol_cal_segmentLen_355": 40,
    "depol_cal_smoothWin_355": 8,
    "isFR": [1, 1, 1, 1, 1, 1, 1, 1, 0, 0], 
    "isNR": [0, 0, 0, 0, 0, 0, 0, 0, 1, 1], 
    "is532nm": [0, 0, 0, 0, 1, 1, 0, 0, 1, 0], 
    "is355nm": [1, 1, 0, 0, 0, 0, 0, 0, 0, 0],
    "is1064nm": [0, 0, 0, 0, 0, 0, 0, 1, 0, 0],
    "isTot": [1, 0, 0, 0, 1, 0, 0, 1, 1, 0],
    "isCross": [0, 1, 0, 0, 0, 1, 0, 0, 0, 0],
    "is387nm": [0, 0, 1, 0, 0, 0, 0, 0, 0, 0],
    "is407nm": [0, 0, 0, 1, 0, 0, 0, 0, 0, 0],
    "is607nm": [0, 0, 0, 0, 0, 0, 1, 0, 0, 1],
[... 94 lines cut ...]
    "yLim_WVConst": [0, 20],
    "yLim_depolConst_355": [0, 0.2],
    "yLim_depolConst_532": [0, 0.2],
    "zLim_att_beta_355": [0, 15],
    "zLim_att_beta_532": [0, 5],
    "zLim_att_beta_1064": [0, 2],
    "zLim_quasi_beta_532": [0, 2],
    "zLim_quasi_beta_1064": [0, 0.8],
    "zLim_quasi_Par_DR_532": [0, 0.4],
    "zLim_FR_RCS_355": [1e-2, 15],
    "zLim_FR_RCS_532": [1e-2, 15],
    "zLim_FR_RCS_1064": [1e-2, 15],
    "zLim_NR_RCS_355": [1e-2, 2],
    "zLim_NR_RCS_532": [1e-2, 2],

    "calibrationDB": "arielle_calibration.db",
    "logbookFile": "/data/level1a/polly/logbook_polly_Arielle_til_2018.csv"
}

```

Output
```txt
flagCorrectFalseMShots,1,flagFilterFalseMShots,0,flagForceMeasTime,0,flagDTCor,1,flagSigTempCor,0,flagLCCalibration,1,flagDepolCali,1,flagWVCalibration,1,flagUsePreviousDepolCali,1,flagUsePreviousWVconst,1,flagUsePreviousLC,1,flagUseManualRefH,0,flagUseSameRefH,0,flagAutoscaleRCS,1,flagMolDepolCali,0,flagTransCor,1,MWRFolder,,dataFileFormat,(?<year>\d{4})_(?<month>\d{2})_(?<day>\d{2})_\w*_(?<hour>\d{2})_(?<minute>\d{2})_(?<second>\d{2})\w*.nc,gdas1Site,cabauw,AERONETSite,Cabauw,max_height_bin,3000,first_range_gate_indx,261  261  261  261  261  261  261  261  262  262,first_range_gate_height,78.75,deltaT,30,dtCorModeLabel,[1],Polynomial coefficients stores in netcdf file,[2],nonparalyzable correction,[3],User defined polynomial cofficients,,dtCorMode,2,dt,0           0           0           0           0           0           0           0           0           0     0.97299      1.0117      0.9674      0.9929      0.9843      0.9391      1.0035           1           1           1   0.0035333     -0.0014      0.0023           0      0.0022      0.0063      0.0003           0      0.0029      0.0028 -7.9098e-06      0.0002           0      0.0001      0.0001     -0.0001      0.0001           0           0           0  1.0645e-07           0           0           0           0           0           0           0           0           0  1.4289e-09           0           0           0           0           0           0           0           0           0,bgCorRangeIndx,10  240,mask_SNRmin,1           1           1        0.01           1           1           1           1           1           1,tempCorFunc,[1],1,[2],1,[3],1,[4],1,[5],1,[6],1,[7],1,[8],1,[9],1,[10],1,[11],1,[12],1,[13],1,,init_depAng,19.99,depol_cali_mode,2,depol_cal_time_fixed_p_start,,depol_cal_time_fixed_p_end,,depol_cal_time_fixed_m_start,,depol_cal_time_fixed_m_end,,m [... cut ...] a_Originator_affiliation,,Data_Originator_affiliation_acronym,,Data_Originator_address,,Data_Originator_phone,,Data_Originator_email,,comment,,logbookFile,/data/level1a/polly/logbook_polly_Arielle_til_2018.csv,radiosondeFolder,,calibrationDB,arielle_calibration.db,imgFormat,png,partnerLabel,,prodSaveList,[1],overlap,[2],aerProfFR,[3],aerProfNR,[4],aerProfOC,[5],aerAttBetaFR,[6],aerAttBetaOC,[7],aerAttBetaNR,[8],WVMR_RH,[9],volDepol,[10],quasiV1,[11],quasiV2,[12],TC,[13],TCV2,[14],cloudinfo,,
```

HolgerPollyNet, 2024-08-01:
@cristoferjimenez @ZPYin Thanks a lot! 🥇  I will implement it the next days. Then, together with the PPC commit, we have a full reproducible processing chain!

ulysses78, 2024-08-02:
Looks good. One question concerning the deadtime-correction polynomial matrix (keyword: "dt").... at the moment the matrix is written channel-wise and polynomial-wise all together. This is very hard to read and distinguish in the end. Is it possible to output it with brackets, like: `[polynom 1.channel][polynom 2.channel][polynom 3.channel]` ....?

HolgerPollyNet, 2024-08-02:
Hi all,

i implemented Zhenpings solution to the PPC, see https://github.com/PollyNET/Pollynet_Processing_Chain/tree/Polly_Config_Storing  or the repsective pull request:
[Polly_Config_Storing](https://github.com/PollyNET/Pollynet_Processing_Chain/pull/282)

@ulysses78 @ZPYin @cristoferjimenez if you have a solution for @ulysses78 idea, please feel free to change the code and update it to the branch before we merge it with dev.

HolgerPollyNet, 2024-11-14:
as it is working since a while and nobody had a better idea I close it for now

closed 2024-11-14 by HolgerPollyNet