PollyNET/Pollynet_Processing_Chain #296: Implement near-range and far-range telesope glueing based on signals

HolgerPollyNet, 2024-10-16:
According to the NOA approach as another option.

This could lead to profiles of bsc, ex, and depol combining both telescopes...

mtsichla, 2024-10-31:
I have added HolgerPollyNet to Gluing Repository to access the code. I attach the read me file for the code.
[GLUING CODE INPUT.docx](https://github.com/user-attachments/files/17587228/GLUING.CODE.INPUT.docx)
How do you want to proceed?

mtsichla, 2024-11-14:
Hello @HolgerPollyNet & @ulysses78 

I calculate the gluing region and the normalization factor after averaging the range corrected signal but then I implement these values in each 30s range corrected profile seperately and not in the averaging signal (so the gluing can be shown in a quicklook also).

The input to my code is the range-corrected signal yes and the following variables and constants:

From a raw file:
1) the length of the variable "measurement_time"
2) the length of the dimension "channel"
3) the length of the dimenson "height"
4) the variable "depol_cal_angle"
5) the variable "if_center"
6) the variable "polstate"
7) the variable "telescope"

(The 5, 6 and 7 will define which channels to glue since not all PollyXTs has the same near field channels.)

Constants:

8) the average interval
9) overlap (in bins, not meters) for each gluing channel (for example 80 which corresponds to 600m)
10) first bin (this can be omitted)
11) correlation coefficient threshold for each gluing channel (for example 0.75)
12) signal to noise ratio threshold for each gluing channel (for example 5)
13) the depolarization calibration angle when the polarizer is in the normal position (for example for aky is 0)

If in your code the sigFR is the range corrected signal and we can have the variable information from another part of the code and insert them there I think it will work.

HolgerPollyNet, 2024-11-14:
Hi @mtsichla ,

can you provide a little bit more detail of the variables and constants. E.g. type (array, which dimensions, etc.) and what values are possible. E.g. for telescope it is not 100% clear what is meant. The I can prepare the implementation. And what is the output? The merged signal only?

mtsichla, 2024-11-22:
Hello @HolgerPollyNet,

Yes I believe that the averaged interval should be the same as for the optical products.

For the high resolution you mention I had in mind that we could provide quicklooks similar to the "Calibrated attenuated backscatter coefficient OC" you used to show in the previous version of tropos but for the glued signals now. 
If we are not interested in showing that or if there is another way then I will adjust the code to glue only the averaged signals.

Lets arrange a meeting when we are all available to conclude. 

Can you provide me with @ulysses78 and @HenniHG emails to send a doodle?

Thank you!

ulysses78, 2024-12-10:
Hi Holger,
I have implemented Marias python-script into picasso by using pyrun. The commit of branch "signal_glueing_mira" is here:
https://github.com/PollyNET/Pollynet_Processing_Chain/commit/d59991ab3167a4861acde6b7d14f9602770babfe

I would suggest to test this branch on the rsd2-server before merging into dev-branch.

ulysses78, 2024-12-11:
I just tested it on rsd2 and there are still some things to take care of ;-)

HolgerPollyNet, 2024-12-11:
okidoki, full trust in you ;-)

ulysses78, 2024-12-11:
Now it runs the whole way through with some new variables: `sigGL[wavelength]` of `dimension[len(profiles),len(heights)]`. But still there is no netcdf-file written out with the glued signals.
At the moment it is not optional to merge NR+FR signals, but will be done automatically within the overlap-correction calculations.