PollyNET/Pollynet_Processing_Chain #303: Channel naming convention

martin-rdz, 2024-11-21:
Can we please come up with an guideline on channel naming for variables in programms and data?
At least to reduce the variaty a bit (532NR, NR_532t, 1064s, 532_cross, 355x, 532T)

Based on a brief exchange with @HolgerPollyNet and @ulysses78 we discussed something like:

### Definition
```
{wavelength}{RR}_{total|cross|parallel}_{NR|FR}
```

- RR (rotational raman) only if applicaple, otherwise omit.
- Theoretically total and FR can also be omitted.
- Vibrational raman channels are denoted as they are (407, not 355H2OVR)

### Application Examples
- 532_total_FR
- 532_cross_NR (not DFOV)
- 1064RR (not 1058)

### Some reasoning behind
- Lower dash instead of Minus to be used in programming code
- near-range and far-range is more correct than near-field, which has a wider field of view
- 1064RR makes more sense than 1058, according to Holger, otherwise it would also be 530

### Open questions
- Abbreviate total, cross, parallel to t, c, p?
- Cross could also be x or s?
- Omit t, FR?

Any thoughts or opinions?
-Martin

cristoferjimenez, 2024-11-21:
Sounds good to me

ulysses78, 2024-11-22:
Here is the full list:

- 355_total_FR
- 355_cross_FR
- 355_parallel_FR
- 355_total_NR
- 355_total_RR
- 387_total_FR
- 387_total_NR
- 407_total_FR
- 407_total_NR
- 532_total_FR
- 532_cross_FR
- 532_parallel_FR
- 532_total_NR
- 532_cross_NR
- 532_cross_parallel
- 532_total_RR
- 607_total_FR
- 607_total_NR
- 1064_total_FR
- 1064_cross_FR
- 1064_parallel_FR
- 1064_total_NR
- 1064_total_RR

Did I forgot something, or does any entry does not make sense (e.g. total_RR)? I would say not to use the abbrevation c,p,t because in this way we can use the whole entry as a title e.g. for plotting.

HolgerPollyNet, 2024-11-22:
I have to say, that I prefer short names, i.e. t instead of total, c for cross etc. but I can survive with everything ;-). What about the DFOV depol channel? It could be classified as near-range depol, but in fact its a different telescope.
How to handle fluorescence in future?

My proposal :
Here is the full list:
wavelength_method_telescope

    355_total_FR ok
    355_cross_FR ok 
    355_parallel_FR ok 
    355_total_NR ok
    355_total_RR --> 355_RR_NR + 355_RR+FR
    387_total_FR --> 387_FR
    387_total_NR --> 387_NR
    407_total_FR --> 407_FR
    407_total_NR --> 407_NR
    532_total_FR ok
    532_cross_FR ok
    532_parallel_FR ok 
    532_total_NR ok
    532_cross_NR ok -_> currently not extisting, or do we consider the DFOV this channel 
    532_cross_parallel --> is in fact a total channel ;-) --> delete
    532_total_RR --> 532_RR_FR + NR
    607_total_FR--> 607_FR
    607_total_NR--> 607_NR
    1064_total_FR
    1064_cross_FR
    1064_parallel_FR --> not yet exiisting
    1064_total_NR
    1064_total_RR--> not existent--> 1064_RR_FR
    **466_FR--> fluorescence
    355_DFOV
    532_DFOV
    1064_DFOV**

What about omitting total?
Something else?

Moritz-TROPOS, 2024-11-22:
Hi,
Sorry, probably, I haven't submit my comment which I wrote this morning.
532_cross_parallel does not make sense.
1064_total_RR and 355_total_RR - as the last two letters usually correspond to NR or FR, it is confusing to read RR at this point (which range might it be?) --> so maybe we should go to 1058_total_FR instead as we have done with the other wavelengths (387 and 607).
Or annother idea: We may use total/cross/parallel only for the elastic channels and use instead "Raman" or "fluorescence" for the other channels, like 607_Raman_FR and 466_fluorescence_FR. And then we could also use 1064_Raman_FR (or 1064_rotRaman_FR). We won't detect cross and parallel other than for elastic channels and so the information "total" is not really necessary for Raman channels.

martin-rdz, 2024-11-23:
The DFOV issue would with `wavelength_method_telescope` be
532_cross_DFOV
i.e. the three telescopes are denoted FR, NR, DFOV
or in the nomenclature above `{wavelength}_{total|cross|parallel|rotraman}_{NR|FR|DFOV}`.

Also I want to highlight the option to omit tags if they are clear.
I.e. 532_total_NR can be abbreviated to 532_NR
And 387_total_FR can also be abbreviated with 387_FR

Finally a thought about maximum brevity discussion, if the specification is clear, one could also replace a c with cross when doing the labels, but the order in the strung must be clear.

ulysses78, 2024-11-25:
The "532_cross_parallel" was a copy paste typo ;-)

ulysses78, 2024-11-25:
One question concerning the DFOV... the present DFOV uses a near-range mirror. Will there be a DFOV for the far-range in the future as well?
The nomenclature above `{wavelength}_{total|cross|parallel|rotraman}_{NR|FR|DFOV}` sounds good to me. But if we use this one, then we should keep the "total" - to have a well defined and structured nomenclature. In this way the grepping is much easier, when used inside the coding.

ulysses78, 2024-11-25:
Does anyone know, if it affects the processing chain, if we just change the channel labels within the polly-config files?

HolgerPollyNet, 2024-11-26:
> Does anyone know, if it affects the processing chain, if we just change the channel labels within the polly-config files?

Is this related to this topic or a general question?

ulysses78, 2024-11-27:
It is related to this topic.

HolgerPollyNet, 2024-11-27:
> It is related to this topic.

Then only @ZPYin can easily reply to this.

ulysses78, 2024-11-27:
Because in the polly config file we have to set this channel naming and nowhere else.

Moritz-TROPOS, 2024-11-27:
To my point of view, it is not just the channel naming in the config file, but throughout the Matlab code, we define channels flags (e.g., flagChannel532FR, or flag532t). These flags are often defined newly and with slightly changing names for the same channel. Having a naming convention would also make these channel flags more uniform.