PollyNET/Pollynet_Processing_Chain #305: Allow PPC to run, in case mergin is not possible

HolgerPollyNet, 2024-12-19:
In case the merging script is not working, currently the PPC is giving an error and no processing is possible.
However, in this case the PPC should just proceed without merging, e.g. with multiple files per day.

Also the merging script currently fails if variables are not available, like death time correction, which is the case for many old data. In this case, this variable should be ignored, so that merging is still possible.

ulysses78, 2024-12-19:
Do not forget the option for `--force_merging true`, when executing `picasso_go24.sh`.

ulysses78, 2024-12-19:
But I see your point. So, if merging with picasso_go24.sh does not result in a merged file, all individual files of that day has to be written into the todo-list. After that, the todo-list has to be processed.

ulysses78, 2024-12-19:
At the moment I am not sure how to handle the plotting for those days.

HolgerPollyNet, 2024-12-19:
The same like usual, one plot for each file or all files in one plot, but differently processed 😉